furrtek/portapack-havoc #2: Freeze during soundboard exit

goldcove, 2016-09-19:
The firmware freezes when you try to exit the soundboard. 
Rev 67a3b02

furrtek, 2016-09-23:
Can't reproduce. Maybe RAM init issue, looking into it. Thanks.

Pine-d-Huitre, 2016-09-25:
Pour reproduire le bug il faut : 
- entrer dans le TEDI/LCR
- retourner au menu principal
- entrer dans le soundboard
- retourner au menu principal
- entrer dans le TEDI/LCR
- retourner au menu principal
- entrer dans le soundboard
et là, en voulant sortir du soundboard ça freeze

furrtek, 2016-09-27:
Still can't reproduce, guessing it's either RAM init (couldn't find anything for now) or something to do with the number of wav files found for the soundboard.

Pine-d-Huitre, 2016-09-27:
Indeed, I can't reproduce it anymore.

furrtek, 2016-12-26:
Closing as couldn't reproduce even after RAM clear. Also changed stuff in RAM init and files checking which may have fixed the initial problem.

closed 2016-12-26 by furrtek