furrtek/portapack-havoc #4: Firmware error 0x1400_000 : 2C

keldnorman, 2017-01-07:
I am getting a Firmware error 0x1400_000 : 2C after installing the new build - is that a known error ? 

mkdir build
cd build
cmake ../
make baseband
make firmware
make program 

dfu-util --device 1fc9:000c --download ../firmware/hackrf_one_usb_ram.dfu --reset
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available

hackrf_spiflash -w ../build/firmware/portapack-h1-havoc.bin 
File size 804744 bytes.
Erasing SPI flash.
Writing 804744 bytes at 0x000000.

Then I reset the device (unplug and replug USB) and get the error on screen (the top shows the menu bar and Portapack|Havoc + icons )

furrtek, 2017-01-08:
It should not (normally) happen all of a sudden, maybe you had the login option enabled ?
If not, then maybe it's uninitialized RAM (again...). When on the fake error screen, try pressing up-down-left-right then the center button. Then if needed go to settings and disable login.

keldnorman, 2017-01-08:
lol - I did not know that there was that fake error screen - that fixed my problem - it works perfectly now :) Thanks :) :) :) 👍

closed 2017-01-08 by keldnorman

darmoy, 2017-01-10:
I had the same problem when flashing. I was going to open an issue before reading this. It never occurred to me that it was the "play dead" screen all of sudden on startup.
I never put the "play dead" as my login screen. But was put by default after the update.
Removed that, problem fixed for me as well.
Thanks

gh0stshell, 2017-01-10:
I flashed my device last night and had no issues

furrtek, 2017-01-10:
I'm not sure why it does that, I added a "magic flag" which has pretty much no chance of matching an uninitialized RAM value. At least it shouldn't happen again if you update your firmware next time.