furrtek/portapack-havoc #5: Setup > UI > Show splash not selectable

dc1rdb, 2017-01-11:
After flashing latest firmware f399548, the show splash option in UI setup is no longer selectable.

Thanks so much for this great firmware, it is getting better and more mischievous with every commit! Really appreciate your hard work.

furrtek, 2017-01-11:
Thanks for the kind words and the report.
It's an issue with the next focusable element guessing, you can still choose the option by using the touchscreen. I'll get to it in a few hours.

dc1rdb, 2017-01-11:
Touchscreen confirmed. Didn't even think about trying it. My fault. Thanks for your quick response.

furrtek, 2017-01-17:
Moved widgets so it works again, will be fixed in next release.

closed 2017-01-17 by furrtek

dc1rdb, 2017-01-17:
Works like a charm - perfect, thanks!