furrtek/portapack-havoc #6: External Reference

Giorgiofox, 2017-01-15:
Hello,
in Capture mode, or in Audio Trasmission, is possible to use external 10 MHz reference?
I'm trying with or without but I cannot notate difference. In HackRF mode, no problem

furrtek, 2017-01-31:
Hey,
It's certainly possible, I'll take a look at how it's done in the HackRF firmware and try to integrate it.

TheCranston, 2018-06-06:
@furrtek, Do you have an update on this?  I'm willing to dive in and take a look, but am not sure where to start.  If you can point me in a direction, I'll give it a try.  thanks!

TheCranston, 2018-06-07:
From IRC #HackRF discussion last night:

sharebrained:	Yes, I recall that feature being lost when I was trying to reduce the number of clocks (and spurs) in the system. Auto-switching is now problematic because the LPC4320 runs from an external clock so I can turn off its 12MHz crystal.

So from that I understand Jared dumped the auto-switching code and hard coded the built in OSC.  I'm totally cool with it being a user selectable switch in a menu to turn off and on manually.  But I'm not sure about the LPC4320 clocking issue he brings up.  I'm not familiar enough with the LPC43xx family to grasp it well.

furrtek, 2018-06-12:
I added an attempt to external clock detection in a82dc196391df851e831c04917333456842e7133. There's a new clock icon next to the SD card icon, it should turn green when a valid external clock is detected. The switch is automatic but the mcu is kept running off the clock generator's crystal (is that a bad thing ?).

This isn't tested at all since I currently don't have an easy way to generate a 10MHz signal.

TheCranston, 2018-06-12:
@furrtek I tested the firmware (hash dc5d6fe).  There is a little grey icon between the SD card and the Bias indicator.  An arrow pointing up with an clock signal above it.  However it's not using the external clock, at least in the receive audio app.  I see may local repeater off by the standard 10ppm of the on board crystal.  The TCXO is recognized in hackrf mode with si5351 register 0 reading 0x1.

Transmitting via the microphone app no longer locks up the unit.  

Let me know what debugging I can do for you.  I think my tool chain in ubuntu is not yet correct.  I'm running the arm embedded gcc directly from arm, not the ubuntu packaged stuff.  i rebuilt the firmware myself as a second try and that firmware wouldn't stay running.  ugh.

jboone, 2018-08-05:
I've added external clock support in the parent project (commit sharebrained/portapack-hackrf@01bad2805c633b4d53f59a28ec99f3f16e61c41d). At power-on, it will detect if an external clock input is present. It will then measure the frequency, and if it is approximately 10MHz (+/-1.5%), the system will use the external clock input for all clocking. If you remove the external clock input, the firmware will crash, as it is dependent on the external clock. I may address this in the future by restructuring the system clock scheme, but that has many other repercussions that need to be considered first.

furrtek, 2018-08-06:
@jboone: I'll do an update with your code in the following days. I also got your e-mail, to which I'll reply at the same time. I'm at some friends right now and there's a bit too much agitation for me to be able to write comprehensible text :)