furrtek/portapack-havoc #7: Distorted audio in WFM mode

dc1rdb, 2017-01-17:
Latest firmware 17451d8 produces very distorted audio in WFM mode (FM broadcast band). 
Note: the distortions are not present in the current continuous integration build of ShareBrained's firmware https://github.com/sharebrained/portapack-hackrf/commit/052fd1c407e506fd3e4a4593b06193cc7d31c2c3

furrtek, 2017-01-17:
Indeed, and the CPLD check reports "BAD".
All audio receivers are broken, trying to find out what's going on.

gh0stshell, 2017-01-30:
Tried the new firmware from today built about 20 hours ago and still have the audio distortion, any eta on a fix, this seems to be causing issue when I record a signal and play it back, using quite an old version with no distortion I dont have the same issue.

djpadbit, 2017-02-11:
I think i found the source of the WBFM distortion, it is caused by a recent commit in ShareBrained's firmware https://github.com/sharebrained/portapack-hackrf/commit/05eb694c0a9d29e3098e396b53a7e0fb05fe9b93. From that point on there is the WBFM distortion but if you build the firmware from the commit just before https://github.com/sharebrained/portapack-hackrf/commit/1986d1faca61117a4a1de9b286458d57c59fe697 there isn't any WBFM distortion so we will probably need to wait for sharebrained to fix this problem

furrtek, 2017-02-12:
Thanks @djpadbit, I reverted the WFM related code and it should work again now.

djpadbit, 2017-02-12:
Yes it works again thanks

closed 2017-02-12 by furrtek