furrtek/portapack-havoc #8: Audio TX Soundboard don't work

Giorgiofox, 2017-02-01:
Hi, 
with latest two release the Soundboard hangup when selected. 

Now what I do to transmit Imperial March?! ;)

PS, feature request: SSB mode transmit for sound board. 

thanks
Giorgio

furrtek, 2017-02-01:
Hello,
There was a long filename issue, it's now working on my side. Please let me know if it solved the problem for you.
Will look into SSB transmit :)

Thanks.

Giorgiofox, 2017-02-01:
Now the Soundboard not hangup when selected, it load the wav selection screen, but If I select a wav it not transmit. The TX led don't turn on.

furrtek, 2017-02-01:
Should be good now. I'm good at fixing half of a bug...

Giorgiofox, 2017-02-01:
It's very good now, perfect. Thank you. 
Now I need to go to transmit Imperial March.

closed 2017-02-01 by Giorgiofox