furrtek/portapack-havoc #10: SD Card Utility

gh0stshell, 2017-02-02:
I used the built in utility to format the card and now I get no filesystem. What file format does the card need to be in, could not easily find this out.

gh0stshell, 2017-02-02:
Also when I record using capture or record through an audio received when I go to soundboard and try to place something I recorded it says no files in /wav/

furrtek, 2017-02-02:
There's no utility to format the SD card. The "Wipe SD card" utility simply clears the first 32MBs to 0, as indicated on the warning screen. Only useful as a delete in emergency situations.
Soundboard loads files from the /wav directory. Recorded audio is placed at root. I'll add a folder selection screen soon.

gh0stshell, 2017-02-02:
what about the format the card should be in?

furrtek, 2017-02-02:
FAT32

closed 2017-02-02 by furrtek