furrtek/portapack-havoc #12: Periodic ticks in soundboard

furrtek, 2017-02-02:
Period related to wav file samplerate.
Caused by WAVFileReader::read being slow ?
Wasn't happening when just using a simple File object.

djpadbit, 2017-03-03:
I did some testing with a simple app (just a button) and WAVFileReader::read didn't produce ticks but did in the soundboard, weird. If using a simple File object didn't produce ticks this probably mean that WAVFileReader::read takes longer than a File object ? You should probably reuse a File object then ? I don't know, anyways hopes this helped :D (i probably just said some useless stuff)

furrtek, 2017-03-05:
Thanks @djpadbit, WAVFileReader::read is directly inherited from File::read so it should take the same amount of time. Thought it was because of the progressbar update but it wasn't.
I reduced the FIFO refill size from 1024 bytes to 512 (smaller but more frequent reads) in 446efa8 and the ticks seem to have disappeared ?

djpadbit, 2017-03-05:
The ticks seem to indeed have disappered. Thanks as always for making such an awesome firmware.

closed 2017-03-05 by furrtek