furrtek/portapack-havoc #9: Jamming bandwidth too large

tit0n, 2017-02-02:
I am getting the error message "Jamming bandwidth too large" when trying the default settings (for example DECT).

Also for some reason, I am unable to change the frequencies using the UI.

gh0stshell, 2017-02-02:
I tried some of the wifi ones and got the same thing

gh0stshell, 2017-02-02:
Also should we be able to set freq? Or is it only with the presets?

furrtek, 2017-02-02:
I increased the max range from 4.5M to 24M, all presets should be able to run now.
Using the min/max buttons should now refresh the values correctly.

tit0n, 2017-02-02:
Thanks. All preset work now. 
We can also change the min/max values for Range1 now, but changing values for Range2 and Range3 doesn't work.

gh0stshell, 2017-02-03:
Flashed the new firmware, I can confirm default ranges work and can change only the first range, whether I leave 2 and 3 checked or unchecked it never saves the freq entered.

furrtek, 2017-02-03:
Bad array index, all buttons should work now.

tit0n, 2017-02-03:
Resolved. Thanks.

closed 2017-02-03 by tit0n