furrtek/portapack-havoc #14: Minor Cross App Issue

gh0stshell, 2017-02-03:
Minor issue, when you go to Capture it shows not implemented, when you go to reply it opens capture(looks like the reply screen) if you could fix this real quick I would highly appreciate it.

gh0stshell, 2017-02-03:
Also how can I donate your project, I only use your firmware

furrtek, 2017-02-03:
Fixed, I edited the wrong line...
You can donate through Paypal furrtek@gmail.com if you wish. It's greatly appreciated.

gh0stshell, 2017-02-04:
Thanks will send you something for your trouble.  :) Is all your current code on Git? I went to check for the file to switch the lines myself to save you from some work, but I couldn't find it. I also may have been looking for the wrong file.

gh0stshell, 2017-02-04:
I will trying out the new firmware in 20min. If you need a tester I am happy to be a genea pig. I check for new firmware on Git every day or 2 and try it out right away.

gh0stshell, 2017-02-04:
Lookin Good, let me know about testing. I emailed you so I have your email you send you some cups of coffee for your work. This is the only FW I use.

closed 2017-02-04 by gh0stshell

furrtek, 2017-02-04:
S'alright, that was in ui_navigation.cpp. I should append the date at the end of the .bin I commit, I'm not very well organized and I often break things that worked well when adding stuff...
Thanks for your help :)

gh0stshell, 2017-02-04:
The commit is on the about screen so that works for checking, but is not
updated