furrtek/portapack-havoc #16: About screen does not show actual git commit hash

dc1rdb, 2017-02-04:
Very minor issue: git commit hash shown on "About" screen is outdatet. Stills shows 15f66eb with latest firmware flashed.

gh0stshell, 2017-02-05:
Added commit to pull current revision number automaticly

furrtek, 2017-02-05:
@gh0stshell : Where are you getting GIT_DESC from ?
Did `make clean` and it take the new commit hash. Guessing the compiler doesn't notice when ui_about.hpp changes.

gh0stshell, 2017-02-05:
I did some research on pulling commit number from git in C++ and that came up as a solution, I cannot get the source to build on my Ubuntu box so I could not test it, I get errors when running make

furrtek, 2017-02-05:
Well, GIT_DESC doesn't mean anything so that seems normal.

gh0stshell, 2017-02-05:
What seems normal?

gh0stshell, 2017-02-05:
I did not get to try since I cannot build the source from just pulling the current main branch
https://git-scm.com/docs/git-describe

I also wanted to try GIT_VERSION

http://stackoverflow.com/questions/1704907/how-can-i-get-my-c-code-to-automatically-print-out-its-git-version-hash

furrtek, 2017-02-05:
But those are just arbitrary #define names, they don't mean anything special to the compiler.
GIT_REVISION is defined in the root's CMakeLists.txt file by calling `git log -n 1 --format=%h`

gh0stshell, 2017-02-05:
Clearly we are talking about different issues.

gh0stshell, 2017-02-05:
I have 3 solutions I found that I would like to try, one is new code in the about app, one is using the make file and last one is using the CMake file. I cannot try any as I cannot git clone this repo and get it to build.

furrtek, 2017-02-05:
I'm talking about the Git commit hash being wrong on the about screen, I thought you were talking about the same thing...
Why did you just change the define name in your pull request ?

gh0stshell, 2017-02-05:
The main issue is the git commit version not being updated, but I was talking about how I have possible solutions, but since I cannot get the source to build I cannot test.

I actually deleted it shortly after as I wanted to git clone and build using the solutions I found before just adding code as I am a beginner coder and still learning C/C++ so I wanted to test out the solutions first before commit.

furrtek, 2017-02-05:
I understood that. So we are talking about the same thing.
GIT_REVISION is being updated correctly, it's just the compiler (or cmake) not picking up the change.

gh0stshell, 2017-02-05:
Sorry, one reason why its hard to comm in tickets. Correct. Trying to tinker but failing at 17%. Will work on building issues later.

gh0stshell, 2017-02-06:
Hash now showing as 46482a1

gh0stshell, 2017-02-08:
New build shows b430b1e

closed 2020-05-24 by furrtek