furrtek/portapack-havoc #20: Building From Source

gh0stshell, 2017-02-06:
Getting many errors when using git clone and trying to build from source, worked through most of the errors and now stuck with this output

[  0%] Building CXX object firmware/application/CMakeFiles/application.elf.dir/rf_path.cpp.obj
/home//portapack-havoc/firmware/application/rf_path.cpp: In constructor 'constexpr rf::path::{anonymous}::Config::Config(rf::Direction, rf::path::Band, bool)':
/home//portapack-havoc/firmware/application/rf_path.cpp:99:2: sorry, unimplemented: use of the value of the object being constructed in a constant expression
  }
  ^
/home//portapack-havoc/firmware/application/rf_path.cpp: At global scope:
/home//portapack-havoc/firmware/application/rf_path.cpp:169:49:   in constexpr expansion of 'rf::path::{anonymous}::config_band()'
/home//portapack-havoc/firmware/application/rf_path.cpp:163:25:   in constexpr expansion of 'rf::path::{anonymous}::config_rx_tx((rf::path::Band)0)'
/home//portapack-havoc/firmware/application/rf_path.cpp:156:38:   in constexpr expansion of 'rf::path::{anonymous}::config_amp((rf::Direction)0, ((rf::path::Band)band))'
/home//portapack-havoc/firmware/application/rf_path.cpp:169:49: error: 'constexpr rf::path::{anonymous}::Config::Config(rf::Direction, rf::path::Band, bool)' called in a constant expression
 constexpr ConfigBand config_table = config_band();
                                                 ^
/home//portapack-havoc/firmware/application/rf_path.cpp:169:49: error: 'constexpr rf::path::{anonymous}::Config::Config(rf::Direction, rf::path::Band, bool)' called in a constant expression
firmware/application/CMakeFiles/application.elf.dir/build.make:2211: recipe for target 'firmware/application/CMakeFiles/application.elf.dir/rf_path.cpp.obj' failed
make[2]: *** [firmware/application/CMakeFiles/application.elf.dir/rf_path.cpp.obj] Error 1
CMakeFiles/Makefile2:247: recipe for target 'firmware/application/CMakeFiles/application.elf.dir/all' failed
make[1]: *** [firmware/application/CMakeFiles/application.elf.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

Please advise on what is needed to compile from source, I have all the needed depends and compilers.

furrtek, 2017-02-06:
What's your version of ARM gcc ? I have 6.2.1 20161205.

ablanquart, 2017-02-06:
Can confirm that it builds fine with the latest arm-gcc build (6 2016q4)

gh0stshell, 2017-02-06:
So let me start from fresh, I git clone, moved over the needed lib and toolchain from hackrf, cd to fw and then run make. Maybe starting from git clone will help me see where is fails.

furrtek, 2017-02-06:
See https://github.com/sharebrained/portapack-hackrf/wiki/Building%20Firmware

gh0stshell, 2017-02-06:
So your saying if I do:

git clone https://github.com/furrtek/portapack-havoc.git
cd portapack-havoc
mkdir build
cd build
cmake ..
make firmware

it should just work

ablanquart, 2017-02-07:
You'll need to copy over the toolchain from portapack-hackrf repo, as well as any other files the build says are missing.

gh0stshell, 2017-02-07:
Already have and know that part already from when I was working through errors when I first tried compiling, just confirming the steps listed in the hackrf guide.

gh0stshell, 2017-02-07:
The error above is where it builds at 17% and then fails with the error above, if I try again it just gives the above error.

I will have access to that computer in an hour

gh0stshell, 2017-02-07:
Deleted all, started fresh, after some tinkering it builds but do you all get a ton of warnings during the build process? Also after building I don't see the bin for flashing.

gh0stshell, 2017-02-07:
My end goal to to have portapack-h1-havoc.bin

gh0stshell, 2017-02-07:
Ubuntu 16.04
gcc-arm-embedded_6-2016q4
toolchain from hackrf in havoc fw folder
lib folder in fw folers with source compiled files
cd build
running cmake ..
make firmware

gh0stshell, 2017-02-07:
Built the portpack-hackrf firmware with 1 warning and have the bin to flash

trying something with portapack-havoc

gh0stshell, 2017-02-07:
following the same steps I used on hackrf I got stuck at 97%, still working on it

gh0stshell, 2017-02-07:
stuck on this error at 97%, not sure what I am missing

cat: hackrf.img: No such file or directory
cat: terminator.img: No such file or directory
firmware/baseband/CMakeFiles/baseband.dir/build.make:78: recipe for target 'firmware/baseband/baseband.img' failed
make[3]: *** [firmware/baseband/baseband.img] Error 1
make[3]: *** Deleting file 'firmware/baseband/baseband.img'
CMakeFiles/Makefile2:353: recipe for target 'firmware/baseband/CMakeFiles/baseband.dir/all' failed
make[2]: *** [firmware/baseband/CMakeFiles/baseband.dir/all] Error 2
CMakeFiles/Makefile2:103: recipe for target 'firmware/CMakeFiles/firmware.dir/rule' failed
make[1]: *** [firmware/CMakeFiles/firmware.dir/rule] Error 2
Makefile:118: recipe for target 'firmware' failed
make: *** [firmware] Error 2

I cloned the havoc repo
made build folder
copied toolchain from hackrf repo cloned
ran cmake ..
make firmware and get tons of warnings and it ends with the above error

gh0stshell, 2017-02-07:
Example of one of the warning I get using the working steps on building hackrf on havoc:

/home/portapack-havoc/firmware/common/bch_code.cpp:349:17: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (i = 0; i < (m + 1); i++) {
               ~~^~~~~~~~~
[ 30%] Building CXX object firmware/application/CMakeFiles/application.elf.dir/bht.cpp.obj
/home/portapack-havoc/firmware/application/bht.cpp: In function 'std::__cxx11::string gen_message_ep(uint8_t, size_t, uint32_t, uint32_t)':
/home/portapack-havoc/firmware/application/bht.cpp:31:103: warning: unused parameter 'relay_state_B' [-Wunused-parameter]
 std::string gen_message_ep(uint8_t city_code, size_t family_code_ep, uint32_t relay_state_A, uint32_t relay_state_B) {

gh0stshell, 2017-02-07:
Narrowed it down to something in the FW folder, I moved over the toolchain from the hackrf repo, is this correct? Am I missing anything?

ablanquart, 2017-02-07:
That's how I do it.  Not sure if that's the "official" way, but I build the portapack-hackrf firmware, then copy over the toolchain, hackrf.img, and terminator.img over to their respective locations in the portapack-havoc directories.  Proceed to build firmware as usual.   You're not really going to build anything different than the portapack-h1-havoc.bin that comes in the firmware dir, however; I usually just flash this file directly to the HackRF.

gh0stshell, 2017-02-07:
The point to me struggling to build by source is so I can tinker and learn from the code, otherwise I would just keep downloading the bin, thanks for the info, I think thats what I was missing and did not know I needed to do that since its not listed anywhere, i will give it a try now, where did you find the hackrf.img terminator.img?

ablanquart, 2017-02-07:
Those files will be in portapack-hackrf/firmware/baseband/ after you've built the portapack-hackrf firmware successfully.

gh0stshell, 2017-02-07:
Do you get a ton of warnings when building havoc vs hackrd that only gives me 2-3?

furrtek, 2017-02-07:
Warnings aren't important, they're just because of -Weffc++ and variables not clearly initialized.
I have to add a bunch of { } and { 0 } to shut them up.

`sudo make firmware` should generate the .img and the final `portapack-h1-havoc.bin` after a bunch of warnings, without doing anything special.

gh0stshell, 2017-02-08:
not for me, just wiped and started fresh again and used sudo make firmware and get the same error as before

~/portapack-havoc/build$ sudo make firmware
Scanning dependencies of target bootstrap.elf
[  0%] Building C object firmware/bootstrap/CMakeFiles/bootstrap.elf.dir/bootstrap.c.obj
[  0%] Building ASM object firmware/bootstrap/CMakeFiles/bootstrap.elf.dir/startup_ARMCM4.S.obj
[  0%] Linking C executable bootstrap.elf
[  0%] Built target bootstrap.elf
Scanning dependencies of target bootstrap
[  1%] Generating bootstrap.bin
[  1%] Built target bootstrap
[  1%] Generating hackrf_cpld_data.hpp, hackrf_cpld_data.cpp
[  1%] Generating portapack_cpld_data.cpp
Scanning dependencies of target application.elf
---------------------------------------------------------------
Goes through and just shows all the warning
---------------------------------------------------------------
Scanning dependencies of target baseband
[ 92%] Generating baseband_ais.bin, baseband_ais.img
[ 93%] Generating baseband_am_audio.bin, baseband_am_audio.img
[ 93%] Generating baseband_capture.bin, baseband_capture.img
[ 93%] Generating baseband_ert.bin, baseband_ert.img
[ 94%] Generating baseband_fsktx.bin, baseband_fsktx.img
[ 94%] Generating baseband_pocsag.bin, baseband_pocsag.img
[ 94%] Generating baseband_nfm_audio.bin, baseband_nfm_audio.img
[ 94%] Generating baseband_tpms.bin, baseband_tpms.img
[ 95%] Generating baseband_wfm_audio.bin, baseband_wfm_audio.img
[ 95%] Generating baseband_wideband_spectrum.bin, baseband_wideband_spectrum.img
[ 95%] Generating baseband_ook.bin, baseband_ook.img
[ 95%] Generating baseband_jammer.bin, baseband_jammer.img
[ 96%] Generating baseband_audio_tx.bin, baseband_audio_tx.img
[ 96%] Generating baseband_afsk.bin, baseband_afsk.img
[ 96%] Generating baseband_replay.bin, baseband_replay.img
[ 97%] Generating baseband_tones.bin, baseband_tones.img
[ 97%] Generating baseband_rds.bin, baseband_rds.img
[ 97%] Generating baseband_ads.bin, baseband_ads.img
[ 97%] Generating baseband.img
cat: hackrf.img: No such file or directory
cat: terminator.img: No such file or directory
firmware/baseband/CMakeFiles/baseband.dir/build.make:79: recipe for target 'firmware/baseband/baseband.img' failed
make[3]: *** [firmware/baseband/baseband.img] Error 1
make[3]: *** Deleting file 'firmware/baseband/baseband.img'
CMakeFiles/Makefile2:356: recipe for target 'firmware/baseband/CMakeFiles/baseband.dir/all' failed
make[2]: *** [firmware/baseband/CMakeFiles/baseband.dir/all] Error 2
CMakeFiles/Makefile2:103: recipe for target 'firmware/CMakeFiles/firmware.dir/rule' failed
make[1]: *** [firmware/CMakeFiles/firmware.dir/rule] Error 2
Makefile:118: recipe for target 'firmware' failed
make: *** [firmware] Error 2

gh0stshell, 2017-02-08:
@ablanquart, thank you for your help with building this repo, appreciate it very much, thanks to you I am good to go with your steps and made a little shell script to automate the steps for me. Thanks!

@furrtek, if sudo make firmware did work I would wonder why with the hackrf repo, since you pulled the initial code from their why sudo would have to be used since hackrf does not, but sudo make firmware as shown above does not work

I am all set for building and can just pull new changes and build now that everything is setup, ill let you get to this issue when you can, I would guess there is only a handful of us building from source and all others are downloading the FW from github.com like I was.

furrtek, 2017-02-08:
Doing `make clean` and `sudo make firmware` I have:
`[ 93%] Generating terminator.img
[ 93%] Generating baseband_ais.bin, baseband_ais.img
[ 94%] Generating baseband_am_audio.bin, baseband_am_audio.img
[ 94%] Generating baseband_capture.bin, baseband_capture.img
[ 94%] Generating baseband_ert.bin, baseband_ert.img
[ 95%] Generating baseband_fsktx.bin, baseband_fsktx.img
[ 95%] Generating baseband_pocsag.bin, baseband_pocsag.img
[ 95%] Generating baseband_nfm_audio.bin, baseband_nfm_audio.img
[ 95%] Generating baseband_tpms.bin, baseband_tpms.img
[ 96%] Generating baseband_wfm_audio.bin, baseband_wfm_audio.img
[ 96%] Generating baseband_wideband_spectrum.bin, baseband_wideband_spectrum.img
[ 96%] Generating baseband_ook.bin, baseband_ook.img
[ 96%] Generating baseband_jammer.bin, baseband_jammer.img
[ 97%] Generating baseband_audio_tx.bin, baseband_audio_tx.img
[ 97%] Generating baseband_afsk.bin, baseband_afsk.img
[ 97%] Generating baseband_replay.bin, baseband_replay.img
[ 98%] Generating baseband_tones.bin, baseband_tones.img
[ 98%] Generating baseband_rds.bin, baseband_rds.img
[ 98%] Generating baseband_ads.bin, baseband_ads.img
[ 98%] Generating hackrf.bin, hackrf.img
[ 98%] Generating baseband.img`

hackrf.img and terminator.img should be generated from baseband/CMakeLists.txt lines 404 and 416. Can't see why it wouldn't.

gh0stshell, 2017-02-08:
I will try adding make clean and see what happens, it did not mention
having to use make clean at any point of the hackrf build guide

furrtek, 2017-02-08:
You shouldn't have to, but diverting from instructions is helpful sometimes.

furrtek, 2017-02-11:
Missing .cmake file added, and dependencies for baseband.img changed.
I cleared everything and cloned again, build still gave a lot of warnings but finished without errors.

gh0stshell, 2017-02-20:
Sorry, went on a week vacation with no connectivity

deleted repo and started fresh

everything works with no workarounds, most errors in the 70ish% range, some in the first 20%

closing now that everything works

closed 2017-02-20 by gh0stshell