furrtek/portapack-havoc #21: SD Card Undected After Using Wipe Tool

gh0stshell, 2017-02-07:
Once I used the wipe tool it wiped the whole card, deleted the partition and put it into a RAW format, once this happened the device would not see the card and I had to use a computer to reformat the card so I could use it again, that was why I was asking about the format it wanted.

furrtek, 2017-02-07:
What do you want to know ? Why did you make a new ticket ?
I already wrote that wipe != format. Wipe = clear the card to make it unreadable in emergency situations. Format it back to what it was originally (FAT32).

gh0stshell, 2017-02-07:
Yes you already told me the format, I opened a new ticket because I did not know or see anything about it suppose to be an emergency full wipe, you only mentioned it wipes the first 32mb of the card or at least thats what the prompt says. I would have no opened a ticket if I knew it was suppose to do this.

gh0stshell, 2017-02-07:
I do not see anything on this page about the wipe tool or it being an emergency all wipe, would have not opened a ticket if this was known

https://github.com/furrtek/portapack-havoc

gh0stshell, 2017-02-07:
The confirm message is misleading if it removes the data and partition of the card

closed 2017-02-07 by gh0stshell

furrtek, 2017-02-07:
Will make that clearer, even if "Wipe first 32MB of SD card" pretty much tells what will happen.

gh0stshell, 2017-02-07:
This to me does not mean it will format the entire card to RAW, wipe can mean different things to different people

gh0stshell, 2017-02-07:
You can change the message to say Format Card or not, I would just suggest putting something on the page you call your wiki, I looked their first before opening a ticket and did not see anything therefore I though I found an issues with that app

furrtek, 2017-02-07:
It doesn't format the card...