furrtek/portapack-havoc #23: Whistle (and Morse Beacon) don't TX

Giorgiofox, 2017-02-09:
In "Whistle" screen the TX button don't work. Is also no possible to set a frequency. 
Also in Morse Beacon Screen I can't TX: maybe I'm missing some file on SD?

furrtek, 2017-02-09:
Whistle and Morse beacon aren't done yet. That's what happens when I add the menu item before finishing the app :/

Giorgiofox, 2017-02-12:
Maybe consider also a simple cw carrier generator: very very useful as signal generator for all lab purpose.

furrtek, 2017-02-14:
Morse TX works now. Working on Whistle.

Giorgiofox, 2017-02-14:
IT WORK! very useful to use it as beacon, compliments and thank you!

furrtek, 2017-02-15:
Just added a super simple CW generator (Utilities menu). Will continue adding on morse TX.
:)

closed 2017-02-15 by furrtek

Giorgiofox, 2017-02-15:
Do you think will be possible to add a sort of cw keying using for example
a gpio or the central button?
Imagine that this can permit to use hackrf and Portapack to make qso @2.4
Ghz without using transverter
and using a simple CW Keyer.
What you think about ?