furrtek/portapack-havoc #24: HackRF release 2017.02.1

Giorgiofox, 2017-02-11:
Hello, 
please integrate hackrf release 2017.02.1 in Havoc as HackRF mode
https://github.com/mossmann/hackrf/releases

furrtek, 2017-02-12:
Hi,
Tried using the new hackrf_one_usb.dfu file, it allows to flash the firmware correctly but "HackRF mode" doesn't work. Nothing in dmesg either. Will look into it.

Giorgiofox, 2017-02-14:
Yes I confirm. Also no "USB" led, so no USB connection.

jboone, 2017-02-14:
If I were to guess, I would say the changes to power management may be tripping up the process of rebooting the H1 into HackRF mode -- perhaps turning off the SPIFI peripheral before the M0 is done with the rebooting process. See this HackRF commit for more details: https://github.com/mossmann/hackrf/commit/2163ebac9c49627692e6a4b1cf99972bab415ba7

jboone, 2017-02-22:
My prior comment has proven to be a poor guess, as I integrated the 2017.02.1 firmware into my repo (commit 66d66b57ab8ef89e00a3491532af8c620f20a9e8) and it works fine. So ignore my lame guess... The issue must be something else.

furrtek, 2017-03-03:
I tried just using the new HackRF .bin but the problem remains.
Actually, there's the same problem even when using an earlier binary: apparently everything goes smoothly up to m4_init, but no signs of life after that.
I checked both image_tag_hackrf and m4_code_hackrf values, they're unchanged. I did change some stuff in the binary image layout to add more baseband modules, maybe that's the cause.

oxoocoffee, 2017-04-01:
Hello,
After building latest version I did try to install it. I gem getting error 0x02401200

1. Enter DFU
2. dfu-util --device 1fc9:000c --download hackrf_one_usb.dfu --reset
3. Reboot and enter DFU mode again
4. make program
5. reboot

At this point I got main screen. I selected Play Dead and I got Firmware error.
Following Wiki I did try to press up, down, left, right and center but that does nto do anything.
Powering off and on bring same screen but each time Firmware Error hex value is different.
I did try to repeat reprogramming from beginning and I always get firmware error :) Fun Fun

Any Ideas?

furrtek, 2017-04-01:
Grrr, I have to find a way to init Play Dead correctly after updates.
Up-down-left-right-center in sequence doesn't show "Please reset" ?

If not, try editing firmware/application/ui_navigation.cpp, uncomment line 414 (`portapack::persistent_memory::set_playdead_sequence(0x8D1);`), rebuild and flash again.

oxoocoffee, 2017-04-01:
I did one see "Please reset" After I installed firmware for first time. Then I did see on reboot full menu and I selected Play Dead, I got error and did try to press up,down,... and I did see Please reset. I did try to press center wheel again but nothing happened. power down/up. FRom this time on I got Firmware error each time on start and up,down,.. does not show Reset.

While writing this up I did uncomment  line you asked me and I did make clean, make firmware

1.Enter dfu
2. dfu-util --device 1fc9:000c --download hackrf_one_usb.dfu --reset
3. Enter dfu
4. make program

Here is output from and below is output from make program

 $ dfu-util --device 1fc9:000c --download hackrf_one_usb.dfu --reset
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download	[=========================] 100%        21992 bytes
Download done.
dfu-util: unable to read DFU status after completion
dfu-util: can't detach
Resetting USB to switch back to runtime mode

And here is output from  "make program"

(master) $ make program
[  0%] Built target bootstrap.elf
[  1%] Built target bootstrap
[ 55%] Built target application.elf
[ 55%] Built target application
[ 82%] Built target baseband_shared
[ 82%] Built target baseband_wideband_spectrum.elf
[ 83%] Built target baseband_wfm_audio.elf
[ 84%] Built target baseband_fsktx.elf
[ 84%] Built target baseband_ert.elf
[ 84%] Built target baseband_audio_tx.elf
[ 85%] Built target baseband_mic_tx.elf
[ 85%] Built target baseband_ads.elf
[ 85%] Built target baseband_jammer.elf
[ 86%] Built target baseband_am_audio.elf
[ 87%] Built target baseband_afsk.elf
[ 88%] Built target baseband_tones.elf
[ 89%] Built target baseband_capture.elf
[ 90%] Built target baseband_pocsag.elf
[ 90%] Built target baseband_rds.elf
[ 90%] Built target baseband_ais.elf
[ 90%] Built target baseband_nfm_audio.elf
[ 91%] Built target baseband_no_operation.elf
[ 91%] Built target baseband_ook.elf
[ 92%] Built target baseband_replay.elf
[ 92%] Built target baseband_sstvtx.elf
[ 92%] Built target baseband_tpms.elf
[100%] Built target baseband
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download	[=========================] 100%        20484 bytes
Download done.
dfu-util: unable to read DFU status after completion
File size 847104 bytes.
Erasing SPI flash.
Writing 847104 bytes at 0x000000.
[100%] Built target program

Let me know if you need anything.. I am here

furrtek, 2017-04-01:
Do you still get the Play Dead screen after uncommenting that line ?
If you still see it, you should be able to do up-down-left-right-center and reset (no power down/up). When in the main menu, go to setup -> UI and make sure "login with play dead" is unchecked. The screen shouldn't show up anymore after that.

I'm sorry, this is probably still caused by uninitialized RAM. Maybe I should just remove the play dead mode while learning how to do things correctly...

oxoocoffee, 2017-04-01:
I do not get menu at all. Each time I power down and up I get to Error screen right away.
As I remember I got this Error on very first time firmware install and reboot. So for clarity I will list of events.

1. Build your firmware
2. Reset harckrf dfu  (2017.02.1)
3. Install your firmware (This is very first installation)
4. Reboot and right away I got Error message.
5. I did press up,down... and I did see Reset message. I pressed center button but nothing happened
6. Power down/up and I did see firmware menu
7. I selected Play Dead and got firmware error message
8. Pressing up, down ... did not show Reset 
9. I did power up/down few times and got to above point 8.
10. Did try to reflash and each time I got on boot to Firmware Error message
11. I did reverted back to portapack-h1-firmware.bin and that worked
12. I uncommented line 414 and did clean make
13. dfu-util --device 1fc9:000c --download hackrf_one_usb.dfu --reset
14. make program (both outputs included above)
15. Each time I power on I get Firmware Error message with different hex numbers
16. Up.down,... does not show reset

So at this point I an not able to see main menu

furrtek, 2017-04-01:
I just updated ui_navigation.cpp to disable that mode at startup.

Giorgiofox, 2017-04-01:
I really hope we found a solution for this very soon, is very useful switch between Havoc and HackRF mode

oxoocoffee, 2017-04-01:
Ok So I did rebuild with your latest changes. After reprogramming and reboot I got menu back. I still see Play Dead option. Selecting it I get Same Firmware Error. But this time rebooting brings it back to menu :) Also up, down , does not work to bring Reset

gh0stshell, 2017-04-02:
I have sometime tonight and I will go through my archive of builds and see
if I can locate the one that fully works until I flash the next and it does
not and I hope that will help you locate the issue

gh0stshell, 2017-04-02:
Build from 020172017 has a working hackrd mode, when I flash the next version from 02142017 it is broken, hope this helps

oxoocoffee, 2017-04-02:
Look like like you mean 02172017? I am assuming last commit on that day in hackrf repo? And your latest?
I id try to build and install latest hackrf and yours. Here is what I noticed

Right now I get menu each time I start. If I select Play dead I still get Firmware Error but up,down,... "Please reset" is shown each time. I am assuming it is porer off/on? Since pressing any other button it does not do anything. If this is the case perhaps better message would be "Press reset button". 

Another thing I noticed that when I select last option to go to HackRF mode I get prompt. I select Yes
and the unit "restarts" but it does not enter HackRF mode. From LED's I can say it looks like it is going into DFU mode. Any ideas?

Trying to run gqrx for example it can't connect. 

I will try your suggestion

gh0stshell, 2017-04-02:
​I will have to double check the BIN date, I have BINs all the way back to
2016 before I started building so its the ones you posted on github that
you could download without having to clone and build the repo, since that
ticket I opened on building is when I started building the repo as well as
downloading the BINs you upload to the github repo​

furrtek, 2017-04-06:
Thanks @gh0stshell, that helped. I diff'd both commits and realized I left the inclusion of the HackRF binary commented after searching why terminator.img was missing...

So it took me a month to notice the HackRF binary didn't load because it simply wasn't there.
I will now go take a bath in sulfuric acid.

gh0stshell, 2017-04-07:
Confirmed working perfectly, flash new firmware, switched to hackrf mode and ran some commands to unlock and lock my car and had no issues

oxoocoffee, 2017-04-07:
I just pulled latest and did try building clean by removing everything in build folder. After cmake .. and make firmware I got this

[ 96%] Generating baseband_wfm_audio.bin, baseband_wfm_audio.img
[ 97%] Generating baseband.img
cat: hackrf.img: No such file or directory
cat: terminator.img: No such file or directory
firmware/baseband/CMakeFiles/baseband.dir/build.make:84: recipe for target 'firmware/baseband/baseband.img' failed
make[3]: *** [firmware/baseband/baseband.img] Error 1
make[3]: *** Deleting file 'firmware/baseband/baseband.img'
CMakeFiles/Makefile2:365: recipe for target 'firmware/baseband/CMakeFiles/baseband.dir/all' failed
make[2]: *** [firmware/baseband/CMakeFiles/baseband.dir/all] Error 2
CMakeFiles/Makefile2:103: recipe for target 'firmware/CMakeFiles/firmware.dir/rule' failed
make[1]: *** [firmware/CMakeFiles/firmware.dir/rule] Error 2
Makefile:118: recipe for target 'firmware' failed
make: *** [firmware] Error 2

Strangely enough I did try to build it again this time with 4 jobs (make -j4) and it build fine. I did check output and did not see any output between. So I just want to report this. I will test firmware when I get home.

furrtek, 2017-04-18:
It verks !

closed 2017-04-18 by furrtek