furrtek/portapack-havoc #29: Please make a new release with binary firmware

ZeroChaos-, 2017-02-22:
I'd like to add this to Gentoo and my Pentoo livecd.

Likely after fixing https://github.com/furrtek/portapack-havoc/issues/24 :-)

jboone, 2017-02-22:
Changes were minimal, work for me, and are now committed on my repo's master: 66d66b57ab8ef89e00a3491532af8c620f20a9e8

Hopefully, this makes the translation to HAVOC easy...

jboone, 2017-02-22:
Duplicate of issue #24, FWIW.

ZeroChaos-, 2017-03-24:
@furrtek poke

I see all the awesome work going into this firmware, please find some time to consider this request or at least respond so I know how to proceed.

furrtek, 2017-03-24:
I still can't figure out why the new HackRF fw doesn't start. I'll have to cut out/replace some code and see what's wrong :(