furrtek/portapack-havoc #37: Add both RX / TX mode with MIC transmit

Giorgiofox, 2017-03-13:
Could be very useful have a full transceiver: RX mode and when PTT is pressed switch to MIC TX mode 
This can permit to do full communications between hackrf also @ 6 GHz ;)

furrtek, 2017-03-13:
It's in the works ;)

Giorgiofox, 2017-03-13:
Ah, fantastic. This and SSB mic transmit allow to make very interesting experiment in SHF band using only linear amplifier, and some things also with Es' HailSat2 geostationary satellite that will be launched very soon, that use uplink @ 2.4 GHz

KC5CQW, 2017-04-18:
Perhaps a settings menu check box to disable the RX when only TX is wanted. Maybe useful in stealth modes and/or battery power savings. 

A squelch setting will also be needed. Well on the way to making the HRF/PP a hand held transceiver!

KC5CQW, 2017-04-18:
Maybe the RX idle current vs active RX/demodulation isn't that much. Disabled RX is probably not needed.

F4GEV, 2017-04-21:
+1
very interresting feature 
low power RX/TX , AM, FM, SSB, digital , 6 GHz = WoW

cdtechnics, 2017-12-13:
Would be very nice to use it as 2,4 Ghz portable handheld transceiver!

KC5CQW, 2017-12-13:
Also as a TX/RX for radio control aircraft using the Multi-Protocol project. Connect a joystick to GPIO’s and command a model and RX telemetry back! 

73, Damon

KC5CQW, 2017-12-13:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module

Perhaps applicable to a different issue number...