furrtek/portapack-havoc #39: Sweep Scan

Giorgiofox, 2017-03-22:
*not an issue, but a feature request*

The new firmware 2017.02.01 as a very interesting sweep scan option that can reach up to 7.5 GHz/s ..
Maybe could be very very interesting offering this option on Havoc Firmware to see on display what happen on very large frequency span as spectrum analyzer.

furrtek, 2017-03-23:
I'll have to check if the FFT can be done quickly enough to have a good refresh rate. And also see how to fit that on the screen.

Giorgiofox, 2017-03-23:
About the screen you can consider to use it in landscape mode

furrtek, 2017-03-23:
I could, but if it needs scrolling it'll be slow (the LCD chip can only scroll in portrait mode :( )

zeroXzero, 2018-09-22:
@furrtek Do you think this is possible? If yes where what would the best starting point? (I am still getting familiar with the entire codebase)

vlada79, 2020-01-19:
This would be really nice feature. It does not have to scroll at all, just make it easy to change freq range/zoom in.