furrtek/portapack-havoc #41: Building latest firmware

oxoocoffee, 2017-04-01:
I just did cloned latest I noticed build problm

My system is Linux Mate 18.01 x64, camke 3.7. After running cmake .. in build I got error of missing arm-none-eabi-g++. After installing missing package gcc-arm-none-eabi I did clean build folder and did run cmake .. again. This time no error. 
Now when I did try to run make I got this error:

(master) $ make
[  0%] Generating hackrf_cpld_data.hpp, hackrf_cpld_data.cpp
[  0%] Generating portapack_cpld_data.cpp
Scanning dependencies of target application.elf
[  1%] Building C object firmware/application/CMakeFiles/application.elf.dir/__/chibios/os/ports/GCC/ARMCMx/crt0.c.obj
arm-none-eabi-gcc: error: nano.specs: No such file or directory
firmware/application/CMakeFiles/application.elf.dir/build.make:75: recipe for target 'firmware/application/CMakeFiles/application.elf.dir/__/chibios/os/ports/GCC/ARMCMx/crt0.c.obj' failed
make[2]: *** [firmware/application/CMakeFiles/application.elf.dir/__/chibios/os/ports/GCC/ARMCMx/crt0.c.obj] Error 1
CMakeFiles/Makefile2:247: recipe for target 'firmware/application/CMakeFiles/application.elf.dir/all' failed
make[1]: *** [firmware/application/CMakeFiles/application.elf.dir/all] Error 2

gh0stshell, 2017-04-01:
building the latest pull I do not get that error on Ubuntu

furrtek, 2017-04-01:
Try using https://launchpad.net/gcc-arm-embedded/+download ?

gh0stshell, 2017-04-01:
Looks like gcc-arm is in the Mint repo

https://community.linuxmint.com/software/view/gcc-arm-none-eabi

oxoocoffee, 2017-04-01:
Yup. That was the issue. I uninstalled mint and added launchpad and I was able to build fine

closed 2017-04-01 by oxoocoffee

furrtek, 2017-04-01:
Will add to the wiki that the version/origin of arm-gcc is important.