furrtek/portapack-havoc #42: WFM broken with latest firmware

dc1rdb, 2017-04-07:
Just flashed latest firmware https://github.com/furrtek/portapack-havoc/commit/b61869f1c04eaa43566648a917c5640f2af2f830 and did some quick checks. WFM mode in FM broadcast band seems to be broken, no demodulation.
AM tested in short wave and aviation bands, working fine here.

djpadbit, 2017-04-07:
Yea can comfirm, its the same issue as last time https://github.com/furrtek/portapack-havoc/issues/7

dc1rdb, 2017-04-07:
HackRF CPLD check reports ok this time, though.

djpadbit, 2017-04-07:
this commit seems to distort WFM https://github.com/sharebrained/portapack-hackrf/commit/05eb694c0a9d29e3098e396b53a7e0fb05fe9b93 and it isn't fixed yet.

furrtek, 2017-04-08:
Thanks for spotting the commit. I will look into it shortly and revert back to the SIMD macros if I can't understand what's going on.

furrtek, 2017-04-18:
Reverted to the SIMD macros in a053c0e, no idea why it does that.

closed 2017-04-18 by furrtek

dc1rdb, 2017-04-19:
Latest pre-built version https://github.com/furrtek/portapack-havoc/blob/master/firmware/portapack-h1-havoc.bin does not seem to include commit https://github.com/furrtek/portapack-havoc/commit/a053c0e234ee3373248607864ddf78ae11d10a18, still shows https://github.com/furrtek/portapack-havoc/commit/fbc054ca75d8784799dec4d10562efdf81bf7cd5 on the about screen.

mullwar, 2017-04-19:
Does this resolves weird signal mirroring bug?