furrtek/portapack-havoc #43: hackrf USB mode not works

Sasul, 2017-04-09:
Since the last update the hackrf USB mode no longer works. How can I solve the problem under windows and install a new firmware.

furrtek, 2017-04-09:
Did you try to just re-flash the original PP firmware as described here:
https://github.com/sharebrained/portapack-hackrf/wiki/Updating%20Firmware%20Windows
?

Sasul, 2017-04-09:
Unfortunately this is not possible. Because this must be the hackrf in the USB mode. Exactly this is no more.

furrtek, 2017-04-09:
DFU is a hardware feature, not software. If the hardware isn't damaged, DFU mode should work.
Use dfu-util and upload the original firmware.

Sasul, 2017-04-10:
Thank you, your tip with the DFU Mode has worked. It is all very well again.

closed 2017-04-10 by furrtek