furrtek/portapack-havoc #45: Compilation Issue

F4GEV, 2017-04-12:
I have errors when i compile last Havoc firmware.

----

[ 96%] Generating baseband_tpms.bin, baseband_tpms.img
[ 96%] Generating baseband_wideband_spectrum.bin, baseband_wideband_spectrum.img
[ 96%] Generating baseband_wfm_audio.bin, baseband_wfm_audio.img
[ 97%] Generating baseband.img
cat: hackrf.img: No such file or directory
cat: terminator.img: No such file or directory
make[3]: *** [firmware/baseband/baseband.img] Error 1
make[3]: *** Deleting file `firmware/baseband/baseband.img'
make[2]: *** [firmware/baseband/CMakeFiles/baseband.dir/all] Error 2
make[1]: *** [firmware/CMakeFiles/firmware.dir/rule] Error 2
make: *** [firmware] Error 2

furrtek, 2017-04-12:
It's amazing how I'm able to make problems occure in a loop.
While I find a real solution, you can copy hackrf.img and terminator.img from Sharebrained's repo.

F4GEV, 2017-04-12:
Many thanks,
works with workaround !!

furrtek, 2017-04-18:
Added required files in 62b9094, even if the makefile should generate them. Workaround for now...

closed 2017-04-18 by furrtek

F4GEV, 2017-04-19:
Sorry issue is present with this update