furrtek/portapack-havoc #46: Audio TX / Microphone / transmit immediately

F4GEV, 2017-04-13:
When you use Audio TX => Microphone
if you enter the frequency manualy, when you click on Done, the HachRF transmit signal without power on the red Led.
If you click on Left button and release, the transmit stop.

F4GEV, 2017-04-13:
i close this issue, after test with spectrum analyser, it's an OL leakage from HackRF One card !

furrtek, 2017-04-13:
It shouldn't leak that much. The transmit pause code might have the same issue as the OOK transmitter for morse tx. Just disabling the PA isn't enough I guess.

kpeezy707, 2017-07-31:
Am i suppose to use a certain type of mic to transmit? tried a few i have and none seem to allow me to talk

furrtek, 2017-08-01:
You need a TRRS (3 ring) jack plug with the Left-Right-Mic-Ground arrangement. I believe that's the most common arrangement for headsets. You can also get Y cables for cheap to plug in earphones and a regular mic.

kpeezy707, 2017-08-01:
Ok cool. I ordered a Y cable to give that a try. I did verify that im using that standard that you mentioned but still no luck with these headphones. Could it be a faulty pack possibly?

kpeezy707, 2017-08-04:
Ive done a little testing before posting this, but i cant get the microphone transmitter to pick up my mic. I've tried converters to the Left-Right-Mic-Ground positioning, tried about 5 sets of head phones. Just wondering if i may have a faulty porta-pack? Or what type and brand of headphones do you guys use? I'm over in the US and all the headphones I've tried have failed.  Cant figure it out for the life of me lol. I've seen some Italian guy on Twitter get it working but he wont give up how he got it to work.

kpeezy707, 2017-08-15:
Totally works now, just updated to latest firmware as of today, not sure what was going on but having some fun with it now sitting in traffic lol, thanks alot!

furrtek, 2017-08-15:
Great :) I updated the binary with Sharebrained's fix for the new audio codec chips.

closed 2017-08-15 by furrtek