furrtek/portapack-havoc #49: POCSAG format display cut lines

F4GEV, 2017-04-14:
From SD Card

20170413074757 466175000Hz 1200 RAW: 7A89C197 7A89C197 7A89C197 7A89C197 7A89C197 7A89C197 7A89C197 7A89C197 7A89C197 7A89C197 7A89C197 7A89C197 7A89C197 7A89C197 3D92F933 7A89C197 
20170413074757 07:47 1200 ADDR:1008824 F3
20170413074758 466175000Hz 1200 RAW: 7A89C197 7A89C197 7A89C197 7A89C197 7A89C197 7A89C197 7A89C197 7A89C197 7A89C197 7A89C197 7A89C197 7A89C197 7A89C197 7A89C197 7A89C197 7A89C197 
20170413074808 466175000Hz 1200 RAW: 7A89C197 7A89C197 7A89C197 7A89C197 7A89C197 7A89C197 43D11DF7 86ECBBEE 8B765D81 D9BB102E DA79CBE3 BBA63DC2 DAD460F8 AC1AD35C 85A69D6A E5FB257C 
20170413074808 07:48 1200 ADDR:1111107 F3
20170413074808 07:47:37**Test.F7-140-Perio**
20170413074808 466175000Hz 1200 RAW: F2E3D52A FA61A235 B26CC4C2 DAB36E2C EC3A71FE 868C9F30 B316AAAA CDBB0C86 E9C1A101 B26CC4C2 DAB36E2C EC3A71FE 868C9F30 B316AAAA CDBB0C86 E9C1A101 
20170413074808 **dique01234567890123456789012345678901234567890**

F4GEV, 2017-04-14:
may be in firmware/common/pocsag.cpp file

furrtek, 2017-04-18:
Display on screen shouldn't be chopped as of 3a1e5b8, will still be in logs since the raw data is added after each batch.

closed 2017-04-18 by furrtek

fantuz, 2017-12-06:
Display is messy, no CR/LF.

Also, i noticed German is not well supported, as I guess other charachtersets.

A nice feat would be to be able to configure the max-length (i guess is a per-user preference) and in the same menu the "filter-out", in order to allow more space for deocdinf on the tiny display

furrtek, 2017-12-06:
I will try simulating messages containing line feeds, not getting any right now.

Apparently there's no way of knowing which charset to use in POCSAG messages. I can add a charset selection menu, if really needed.

fantuz, 2017-12-06:
and font size :) 

no, anyway, thanks for all the features you added to this Portapack expansion. It arrived today by post and I flashed it with your HAVOC firm without any hesitation !

cheers