furrtek/portapack-havoc #54: POCSAG : Ignor address does'nt work

F4GEV, 2017-04-19:
When selected with a  test address, the consol display messages => filter does not work.

*Désolé furrtek, une belle release quant même *

furrtek, 2017-04-19:
Oops, what was I thinking ? It was only filtering "address only" messages to avoid duplicate notifications in the console. Now, nothing is displayed and logged at all for the ignored address (commit 40b49e2).

New messages starts with a green header (Time, baudrate, address, function).
Continued messages are in regular white text. The line breaks should be OK on the console.

F4GEV, 2017-04-19:
Works, thanks Monsieur
so in my town the Test Add is 1111107 is it possible to save the last enterred Add in RAM (battery saved" ?

furrtek, 2017-04-21:
Added auto save in 90feadd

F4GEV, 2017-04-21:
all works fine 
Good job

closed 2017-04-21 by F4GEV