furrtek/portapack-havoc #61: Compiling issue on Ubuntu 16.04

mrmookie, 2017-05-14:
I'm having difficulty building the firmware myself. I've attached a screenshot of the error.. 

[image]

I installed the package 'gcc-arm-none-eabi' I'm a bit confused if I have the right one installed. Thanks for any help.

gh0stshell, 2017-05-14:
https://launchpad.net/gcc-arm-embedded/+download

furrtek, 2017-05-18:
The link @gh0stshell gave should give you a version of gcc-arm-none-eabi which doesn't produce that error.

mrmookie, 2017-05-18:
Thanks for the info, I've been about to get further but I seem to have a different error now.. 

[image]

Showing I'm using cmake version 3.8.1 and arm-none-eabi-gcc at 5.4.1.  Am I missing something obvious? Thanks for the help.

furrtek, 2017-05-18:
Those 2 files not being spotted by cmake is still strange...
Do you have hackrf.img and terminator.img in firmware/baseband/ ?
They should be generated automatically in time, and anyways I included them in the repo so they should be there.

mrmookie, 2017-05-18:
Yes, both hackrf.img and terminator.img appear in portapack-havoc/firmware/baseband/

mrmookie, 2017-05-18:
I notice all the path's in /firmware/Makefile reference /home/furrtek... that path obviously doesn't exist on my machine. Could this be causing an issue?

gh0stshell, 2017-05-18:
I had the same thought but after some help it worked even though I did not change the paths

You had the same issue as me at first, try the workaround

 copy hackrf.img and terminator.img from Sharebrained's repo and place them in the correct dir

furrtek, 2017-05-18:
I shouldn't have committed the makefile with my home directory in it... I'll fix that shortly.

mrmookie, 2017-05-18:
Ok so thinking about this... The missing files are in the repo /firmware/baseband/ but during build I think it's looking for the files (which are missing) in /build/firmware/baseband ? Does that help or make any sense? Once I placed the Sharebrained ones into the /build/firmware/baseband directory, the build completed!

gh0stshell, 2017-05-18:
as part of the build did you do mkdir build?

mrmookie, 2017-05-18:
Yes, just confirmed..

I have it working now, you need to copy the hackrf.img and terminator.imp files that are in /firmware/baseband to the build folder /build/firmware/baseband. You don't need to get it from the Sharebrained repo.

Yes, I followed the WIKI and ran mkdir build, cd build, cmake.., make firmware. Then copied the needed files to my build folder.