furrtek/portapack-havoc #71: LoRa RX/TX

korczis, 2017-07-10:
Hi,
can you please implement LoRa TX/RX? This will be really super-cool in upcoming IoT era.

https://en.wikipedia.org/wiki/LPWAN

furrtek, 2017-07-10:
Added to list along with Sigfox :)

korczis, 2017-07-18:
@furrtek great, thanks

fantuz, 2017-12-07:
https://ask.sigfox.com/storage/.../55-trame-geoloc-1205-v12.pdf
for reference

fantuz, 2017-12-08:
I found that useful for on-the-fly decode-> https://www.oemserver.com/tools/OysterSigfox/UplinkDecoder.html
Otherwhise the JS behind it: https://digmat.freshdesk.com/support/solutions/articles/16000062989-decoding-the-oyster-sigfox-payload-javascript-example-code
And also https://digmat.freshdesk.com/support/solutions/articles/16000061215-decoding-the-oyster-sigfox-payload-examples amongst many other...

fantuz, 2017-12-08:
And encoding -> http://community.td-next.com/viewtopic.php?f=5&t=64&p=378#p378

Ops sorry I just realized the topic here was LoRa... apologize.

fantuz, 2017-12-08:
I promise this is the last link here -> https://digmat.freshdesk.com/helpdesk/attachments/16015593787

chudgoo, 2018-02-24:
Having just picked up a ESP32 based Arduino module, I've already had some fun with LoRa.
I flashed the board to be a LoRa transmitter, captured the bursts on the portapack, flashed the again board to be a receiver and then replayed the packets. Works like a charm, in one direction anyway.

I HEARD LoRa SHIT THIS MANY TIMES!  hello 265' with RSSI -99
I HEARD LoRa SHIT THIS MANY TIMES!  hello 260' with RSSI -96
I HEARD LoRa SHIT THIS MANY TIMES!  hello 261' with RSSI -99
I HEARD LoRa SHIT THIS MANY TIMES!  hello 263' with RSSI -99
I HEARD LoRa SHIT THIS MANY TIMES!  hello 264' with RSSI -100
.
.

It would be awesome to have an interface like the POCSAG transmitter where you can set the message text and whatnot... and even more neat to decode.  Fun fun!