furrtek/portapack-havoc #75: Latest hardware version possible compatibility issue

ncoop23, 2017-07-24:
There are changes in the latest hardware revision of the PortaPack which require a firmware update to allow the device to work.  The code is not commented enough for me to understand what exactly is going on or else I would take a crack at fixing it.

ncoop23, 2017-07-24:
It looks like they changed the LCD on the new hardware or at least how the backlight works.

jboone, 2017-07-24:
I'm traveling at the moment, but was hoping to take a crack at merging the necessary changes, since I'm the party responsible for the hardware changes... ;-)

@ncoop23, do you already have a revision 20170522 board in your hands?

        - Jared

ncoop23, 2017-07-24:
I have one of the boards and can test on it.  I'm afraid to flash the new firmware for fear of bricking it since I don't really know what got changed.

jboone, 2017-07-24:
Holy smokes, looks like Furrtek already merged all the changes! I'll verify on my new and old hardware and share my results shortly.

        - Jared

jboone, 2017-07-24:
Furrtek doesn't have the newest changes incorporated, but was up to date a month ago. That's recent enough that HAVOC should run on the new hardware. I checked it out, and while there might be a bit of breakage, there's no harm in running the latest code compiled from source on your 20170522 PortaPack. I will look into merging the newest changes from my repository, they *should* be fairly simple.

ImDroided, 2017-07-24:
I can confirm the firmware works. I have the latest board. Some stuff is having issues it seems but most seems to work.

gh0stshell, 2017-07-24:
Do we have to upgrade boards to keep using Havoc?

jboone, 2017-07-24:
@gh0stshell Nope, software merged from sharebrained/portapack-hackrf will detect the board revision (by seeing which audio codec is on the I2C bus) and load the correct drivers and CPLD code. So old PortaPacks will continue to work with HAVOC. However, I need to send Furrtek a new PortaPack board...

jboone, 2017-07-24:
@ImDroided Your #76 mic issue is probably caused by the HAVOC repo being a bit behind. I fixed some mic configuration stuff for the new hardware revision in the last month's commits. However, even after the merge with the latest code, there may still be issues that need to be resolved. We'll check it out.

Redouan29, 2017-07-24:
So the new hardware revision of portapack works fine and is safe to use havoc addon? Or should we wait?

furrtek, 2017-07-24:
I'll sync HAVOC today with @jboone latest code. I saw some new files related to the new audio codec and LCD I/O but you'll have to test it yourself since I have the 1st board version.
I'll try to make sure the fw always on both versions :)

Redouan29, 2017-07-24:
@furrtek & @jboone thanks!

ImDroided, 2017-07-25:
Ok so I just compiled and flashed your most recent changes and its still a no go on the mic. LCD seems to work a little better.

ImDroided, 2017-07-30:
Mic is now working just fine LCD seems to perform better on my end also.

closed 2017-07-30 by furrtek

jashsu, 2017-08-01:
Will a prebuilt binary release be uploaded?