furrtek/portapack-havoc #76: mic issue

ImDroided, 2017-07-24:
Hi I have just received my portapack the new all black model and I have tried 10 different mics and none of them work with the mic transmit. I hear the roger beep on my radio but no audio. I have tried all types of mics from ear buds to my gaming headsets and I can hear just fine with them just no mic. I have even tried the samsung ear buds that came with my old phone which says is compatible but no luck. Am I using a wrong setting here or is this a bug? My buddy seems to be having the same issue.

jboone, 2017-07-24:
I'm curious what is the small date code on the front of your PortaPack -- 20170522 or 20150901?

        - Jared

ImDroided, 2017-07-24:
20170522. We just got these the other day.

ImDroided, 2017-07-25:
This issue still exists any suggestions?

ImDroided, 2017-07-30:
This is now working please close this issue.

closed 2017-07-30 by ImDroided