furrtek/portapack-havoc #78: Build issue

ncoop23, 2017-07-27:
So I am working to build the firmware and needed to do a couple steps to install the compiler

apt-get install gcc-arm-none-eabi
apt-get install dfu-util

cmake .. works fine

After this I am having an issue where I am unable to build when I run make firmware.  The issue is described below:

On line 117 of ui_scanner.hpp: "array must be initialized with a brace-enclosed initializer"  

Did I install the wrong toolchain or something?

gh0stshell, 2017-07-27:
This the one you used?

https://launchpad.net/gcc-arm-embedded/+download

furrtek, 2017-07-27:
It seems to be a GCC bug. As @gh0stshell wrote, try using a newer version.

ncoop23, 2017-07-28:
I'm in Kali and it's not wanting to take.  I tried updating arm-none-eabi and no luck still.  The full error:

on `make firmware`
```

[ 38%] Building CXX object firmware/application/CMakeFiles/application.elf.dir/ui_navigation.cpp.obj
In file included from /root/portapack-havoc/firmware/application/ui_navigation.cpp:53:0:
/root/portapack-havoc/firmware/application/ui_scanner.hpp:117:42: error: array must be initialized with a brace-enclosed initializer
  std::array<Color, 240> spectrum_row { 0 };
                                          ^
firmware/application/CMakeFiles/application.elf.dir/build.make:3488: recipe for target 'firmware/application/CMakeFiles/application.elf.dir/ui_navigation.cpp.obj' failed
make[3]: *** [firmware/application/CMakeFiles/application.elf.dir/ui_navigation.cpp.obj] Error 1
CMakeFiles/Makefile2:215: recipe for target 'firmware/application/CMakeFiles/application.elf.dir/all' failed
make[2]: *** [firmware/application/CMakeFiles/application.elf.dir/all] Error 2
CMakeFiles/Makefile2:137: recipe for target 'firmware/CMakeFiles/firmware.dir/rule' failed
make[1]: *** [firmware/CMakeFiles/firmware.dir/rule] Error 2
Makefile:131: recipe for target 'firmware' failed
make: *** [firmware] Error 2
root@kali:~/portapack-havoc/build# 

```

It seems like I need to turn off option strict and option explicit or something.  I come from .NET so this is kind of new to me.  I would be happy to update the wiki with the process.

ncoop23, 2017-07-28:
From what I'm reading it seems like it is being picky and wanting something like: 

std::array<Color, 240> spectrum_row = { 0 }

as per: http://en.cppreference.com/w/cpp/container/array

furrtek, 2017-07-29:
I had it build with 6.3.1 20170620 (6-2017-q2-update).
I changed the line to respect C++11 init.

Redouan29, 2017-07-30:
My building of firmware is still failing. Can maybe build it for me? Honestly I have difficulty to install 6.3.1 6-2017-q2-update.

ImDroided, 2017-07-31:
Here is my copy from earlier today. I hold no responsibility for this file. I flashed it to my portapack and it worked. 
[portapack-h1-havoc.zip](https://github.com/furrtek/portapack-havoc/files/1186202/portapack-h1-havoc.zip)

ncoop23, 2017-07-31:
Yeah, basically you need to make a ubuntu vm just to get this latest version of the build chain from the ppa as the kali repo doesn't have the same version in it.

gh0stshell, 2017-07-31:
Ubuntu has the same one until you use that PPA, when building mine at first I installed what was provided, until furrtek sent me the correct one to use it would fail, gcc-arm-embedded works while default one provided in the default Ubuntu and Kali repo gcc-arm-none-eabi fails

ncoop23, 2017-07-31:
I guess it's up to the admin on what coding standard they want to use. My vote is for something like C11 that is widely supported without having to jump through hoops and think that might encourage more developers to make stuff for this but I'm just some random guy trying to make this thing work right and add some features to it for HAM radio use.

Redouan29, 2017-07-31:
@ImDroided Thanks! I already figured out. For building it, just start a fresh Ubuntu 16.04 in your vm and than;

sudo apt-get remove gcc-arm-none-eabi binutils
sudo add-apt-repository ppa:team-gcc-arm-embedded/ppa
sudo apt-get update
sudo apt-get install gcc-arm-embedded

Than:
apt-get install dfu-util

Than:
apt-get install cmake

Now you can build and make firmware

Redouan29, 2017-07-31:
maybe an offtopic question, but I have a question about;

Give it some time. Put your HackRF in DFU mode (keep the reset button pressed and click on the DFU button). Then, upload the DFU update program with dfu-util: from the firmware/ directory, run:

dfu-util --device 1fc9:000c --download hackrf_one_usb_ram.dfu --reset

I never found a hackrf_one_usb_ram.dfu I used from the firmware directory the hackrf_one_usb.dfu after that I flashed it with hackrf_spiflash -w portapack-h1-havoc.bin

fantuz, 2017-12-11:
Hello @Redouan29, me too I was wondering...

I only dfind that: ./firmware/hackrf_one_usb.dfu within HAVOC project.
But can only find ./firmware/build/hackrf_usb/hackrf_usb.dfu within the original HACKRF project.

Sizes are comparable, havent inspected further so far.
Only to avoid misconceptions, I own the HACRF "final built" rather than the HACKRF-one kickstarter....

closed 2020-05-24 by furrtek