furrtek/portapack-havoc #80: Play dead mode

Redouan29, 2017-08-03:
Is it me or is the play dead mode broken?
If I set in the Setup/UI the "login with play dead" on, it is still always turning on normally (on a new power cycle).

djpadbit, 2017-08-05:
I don't think the play dead mode is broken, it just has been disabled (see commit https://github.com/furrtek/portapack-havoc/commit/f6b02c31b5d7c298ec71a3049f97bdf1cc906d00 ). It should work again if you uncomment the code that has been commented out. But then again I might be wrong, hope it helps anyways

furrtek, 2017-08-12:
It was indeed disabled because it kept causing problems after updates. I'll get it working correctly one day.

Redouan29, 2017-08-29:
Ok clear for me. Thanks for the answers!

closed 2017-08-29 by Redouan29

h4waii, 2018-03-05:
Any update on having this enabled again?