furrtek/portapack-havoc #81: Issue with Firmware (git hash aa29348)

nohlemeyer, 2017-08-03:
Hello,
i flashed hackrf with portapack about a few months ago with the havoc-firmware with is really nice so far :)

Today i tried to update the firmware, compiled a new binary, and wanted to flash it with
hackrf_spiflash -w portapack-h1-havoc.bin
but it is not recognized:

hackrf_open() failed: HACKRF_ERROR_NOT_FOUND (-5)

 I tried to start in DFU, but it says:

dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Deducing device DFU version from functional descriptor length
dfu-util: No DFU capable USB device available

dfu-util -l gets me:

dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available
even when i try this after selecting "HackRF Mode" in the Main Mode.

How to upgrade or to restore original firmware?

ImDroided, 2017-08-04:
Have you tried a different cable and port. I was having issues too but it was a bad cable.

nohlemeyer, 2017-08-05:
Today i tried a new cable, and yes, it worked. :-)

closed 2017-08-05 by nohlemeyer

ImDroided, 2017-08-06:
Good. Glad that worked. I don't get it USB is USB but I have a few cables that work fine with one device but not with others....