furrtek/portapack-havoc #83: Frequency Manager not working

ereuter, 2017-08-09:
When I go to Frequency Manager in Utilities, the device freezes.  If I try to save a frequency elsewhere, I get a "No database files" screen with exclamation point.  If I try to load, it freezes the device.  

I tried touching FREQMAN.TXT on the SD card, and no change.  

Hardware is 20170522 version.

furrtek, 2017-08-10:
There's no FREQMAN.TXT file anymore, see sdcard/FREQMAN/ in the repo.

ereuter, 2017-08-10:
Thanks. I was going by the wiki. In any case, from a freshly formatted card, what do I need to do to get the frequency manager running?  Copy the files from the repo to the root?2

ImDroided, 2017-08-10:
copy the sdcard folder to a sd card and everything works.

ereuter, 2017-08-10:
Thanks.  It's working now.  If I missed that in the documentation, I apologize.

furrtek, 2017-08-10:
The wiki wasn't updated, it will be soon :)

closed 2017-08-10 by furrtek