furrtek/portapack-havoc #84: Add recording info in WAV file INFO block

furrtek, 2017-08-11:
Frequency, mode...

furrtek, 2017-11-09:
Added info blocks: "Portapack" as artist and the tuning frequency as title.

xmycroftx, 2017-11-11:
ExifTool Version Number         : 10.10
File Name                       : AUD_0006.WAV
Directory                       : /media/xmycroftx/sd/
File Size                       : 1248 kB
File Modification Date/Time     : 2017:11:10 23:37:36-08:00
File Access Date/Time           : 2017:11:10 23:39:11-08:00
File Inode Change Date/Time     : 2017:11:10 23:37:22-08:00
File Permissions                : rw-r--r--
File Type                       : WAV
File Type Extension             : wav
MIME Type                       : audio/x-wav
Encoding                        : Microsoft PCM
Num Channels                    : 1
Sample Rate                     : 48000
Avg Bytes Per Sec               : 0
Bits Per Sample                 : 16
Artist                          : PortaPack
Title                           : 107300000Hz

furrtek, 2017-11-11:
Thanks ! I was testing it with Audacity, glad to see that another tool also understands the tags.

closed 2017-12-08 by furrtek