furrtek/portapack-havoc #86: ADS-B RX amp/gain settings missing

dc1rdb, 2017-08-12:
ADS-B reception is only possible after first selecting a different receiver mode (e.g. SPEC, WFM...) and setting pre-amp and gains in that mode. Apparently, the default values (pre-amp 0, baseband and if gain 32) are not sufficient to allow frame decoding in my case.

kpeezy707, 2017-08-15:
I know its in dev right now, but how do i use it so far, on the ADS-B Rx screen all i see is a field with a 6 digit number that i can scroll through, is the modes somewhere else?

furrtek, 2017-08-16:
Fixed in 9d902bc224b28e2785680293bdd0a298c40aad70. Pre-amp enabled, LNA and VGA values set at 40 by default.

dc1rdb, 2017-08-16:
Thanks a lot - works fine now, getting lots of frames. Is there anything I can help for further development?

furrtek, 2017-08-28:
Would you happen to know how squawk codes are encoded ? I'm seeing different methods.

closed 2017-08-28 by furrtek

dc1rdb, 2017-08-28:
Hmm, let me check. Have you had a look at the dump1090 code at https://github.com/mutability/dump1090 ?

dc1rdb, 2017-08-28:
http://www.radartutorial.eu/13.ssr/sr24.en.html and
[ICAO Annex 10 Volume 4.pdf](https://github.com/furrtek/portapack-havoc/files/1255614/ICAO.Annex.10.Volume.4.pdf)