furrtek/portapack-havoc #88: Wrong longitude in ADSB RX

furrtek, 2017-08-17:
Sometimes correct. Might be caused by an odd/even frame mix-up.
Latitude and altitude seem about right.

ImDroided, 2017-09-02:
Ok my north and south seem to be correct. My East and west is locating planes in Massachusetts where I am in Illinois near Chicago. If you point me in the right direction as to which files handle this I will take a look if I get some free time.

furrtek, 2017-09-02:
Position decoding is done in decode_frame_pos, in /firmware/common/adsb.cpp

xmycroftx, 2017-11-09:
I started digging on this, and saw some of the flapping back and forth between locations.  Since there wasn't a great way (I tried screencapping but that was a challeng) there's now an adsb.txt log that's being created in the latest builds.  So here's a sample of the failure.  I've redacted the ICAO but that shouldn't affect the rest of the parsing.

`20171109015552 8F999999582384CAF8B2897683F3 ICAO:999999 Alt:6000 Lat37.19 Lon2.28 
20171109015553 8F99999958239136A6046C6B3952 ICAO:999999 Alt:6025 Lat37.20 Lon-53.-50`

 For the relevant bits it looks like the first is the even frame.  I threw this in a quick python test for comparisn in pyModeS and after some poking got: 
`In [33]: adsb.position(even,odd,event,oddt)
Out[33]: (37.8196, -122.48842)`
Which is much more realistic given the location where I grabbed those frames. Even if I mix up the frame order I get a couple of interesting failure mode examples:
`In [31]: adsb.position(odd,even,oddt,event)
Out[31]: (-40.81073, 122.78961) 
In [37]: adsb.position(odd,odd,oddt,oddt)
Out[37]: (1.20943, 2.16435)
`
The second one is close, and if the implementation is using approximations similar to dump1090 I can see how that could be it.   It might be in the CPR parsing itself I'll see if I can take a look.

furrtek, 2017-11-09:
Thanks a lot for the insight. I wasn't sure if it was the mixing up of odd and even frames, or my badly translated CPR conversion code. Now I'm sure :p

xmycroftx, 2017-11-11:
So if two frames arrive in the same second (same timestamp) it will always use the odd frame since the counter isn't as precise as the frame rate Which is probably why it flaps back and forth.  Why it's not getting the right location is still a mystery to me.

KC5CQW, 2018-04-26:
Perhaps some type of voting logic against a reference location set by the user will “force” a latitude. In my case, the Lat. shows around 147E deg but I’m at -92W deg. With a “you are here” location set, then it should be able to understand true and false frames. How is the positive/negative E W directivity currently determined?

KC5CQW, 2018-04-26:
Ok, it looks like a lookup table is needed for the NL function to parse the correct CPR zone. 
See the post reply here:

https://aviation.stackexchange.com/questions/3707/how-does-the-ads-b-compact-position-report-nl-function-work

troussos, 2019-05-21:
Opened a PR for this that should resolve it - https://github.com/furrtek/portapack-havoc/pull/242

Aircorps, 2019-05-27:
Hi all, I'm new to this and just bought the hackrf portapack Havoc version, one of the first things I noticed was longitude is out by 70 to 80 degrees. Go easy on me, I've never compiled anything, only done firmware on radios using simple dfu utility,  I'm not even sure if this unit had Dev support any longer, so please be gentle :) Thanks, Roger

troussos, 2019-05-28:
> Hi all, I'm new to this and just bought the hackrf portapack Havoc version, one of the first things I noticed was longitude is out by 70 to 80 degrees. Go easy on me, I've never compiled anything, [...]

You can find the instructions for building here. You'll need to install some common build tools, but the Ubuntu PPA on that page makes it pretty easy.
https://github.com/furrtek/portapack-havoc/wiki/Building-from-source

After that, there are instructions for flashing here (but it sounds like you're already familiar with that process)
https://github.com/furrtek/portapack-havoc/wiki/Flashing

As long as you build from master - you'll have the update that resolves this issue.

Hope this helps a bit.

furrtek, 2020-01-03:
Fixed in 8fd2d4b1fc31ee0807ef00a5f20a869418e9106e

closed 2020-01-03 by furrtek

PeterDragonTM, 2020-02-23:
Greetings, I am not sure if the issue I see is the same, but it sounds like it. When using the ADS-B, Aircraft Details, commonly will show GPS long and lat in either China or Iraq. The flights I am seeing are in Florida. 

Example Coordinates:
Flight: CONN346 (Based out of Orlando Florida)
30°26'10.0"N 101°54'52.0"E

Does the above fix address this, is the above commit in the current firmware, or do I need to build it from master?