furrtek/portapack-havoc #93: ADS-B wrong map position and freezing unit

ImDroided, 2017-08-28:
After the latest work ADS-b for me is not showing the correct map location puts planes in the middle of the ocean... I live near Chicago which is far from any ocean. Also when I go back from map view and go to select another plane most of the time the unit freezes and I have to hit the reset button. Anything I can do to help just let me know. Awesome work by the way.

furrtek, 2017-08-28:
Thank you :) The map position reflects the possibly wrong longitude values (issue #88), the latitude should be correct. The freezes are hopefully fixed in 986b54e43fd797507348884817892f20736eba19.

ImDroided, 2017-08-30:
ADS-B RX seems to be working fine now.

closed 2017-08-30 by ImDroided