furrtek/portapack-havoc #98: Add TouchTunes Control Support

h4waii, 2017-09-21:
TouchTunes remote runs on 433.92MHz ASK OOK, 256 possible PINs to use from the remote. 

Would be nice to be able to bruteforce, or simply RX from an existing transmission, and provide controls.

This repository has details on an implementation for the Yard Stick One => https://github.com/notpike/The-Fonz

furrtek, 2017-09-22:
Interesting ! Those are jukebox remote controls ?
They're not the most common here, would you be able to test it ?

furrtek, 2017-09-23:
Added under `Transmitters -> TouchTunes remote` in 26949773bb47cfd363c5e904d7cba4eebd68ec5c.
Let me know if you can test it and if it works or not, so that I can add the other buttons.

h4waii, 2017-09-23:
Wow. That was fast! Will check functionality when I can do so next, but that might be a very long time.

Thanks for adding this!

ImDroided, 2017-09-23:
I will also have to try to test this. The place I went to for lunch had this unit.

furrtek, 2017-09-24:
Great. Those aren't the most popular here (never seen one actually), so all I was able to do is record the transmission with my RTLSDR key and see if the data and baud rate match. I converted data from Notpike's script by hand so I hope most buttons work.
Also, there's a scan function but since I don't know how many repeats the receiver needs to accept a frame, I hardcoded 4 and spaced the frames quite a lot. That might take a while to go through all 256 pins.

gh0stshell, 2017-09-24:
Could something like this be added for remotes that use 10 dip switches
with OOK encoding(maybe ASK OOK, but it looks like just OOK)? I can use my
RTLSDR to record my signal, see the 1's and 0's in another audio app and
then push it back using another piece of hardware, I know of one project
that does this, but the code is broken and would be better if part of havoc
so I can use one device for all.

furrtek, 2017-09-24:
See `Transmitters -> OOK remote encoders`. The formats are hardcoded presets, if none match the encoder you want, see /firmware/application/protocols/encoders.hpp or give me a chip reference.

gh0stshell, 2017-09-24:
Will do, I have not upgraded in a bit so I will pull the latest and upgrade first and then do some tinkering and get the FCC ID/CHIP of the remote

furrtek, 2017-09-24:
Listening to the signal in Audacity reminded me of a very similar sound I heard when listing codes for the TV-B-Gone years ago. Turns out TouchTunes used a NEC IR remote encoder !
http://www.sbprojects.com/knowledge/ir/nec.php

ImDroided, 2017-10-06:
Anyone have a chance to test this yet?

furrtek, 2017-10-06:
Same question :)

ImDroided, 2017-10-06:
I will have to go to the ice cream shop that has this this weekend

chudgoo, 2017-10-06:
The only one in my area is at an American Legion (bar for veterans) and they are very change adverse...  maybe some Slayer would liven up the place.  Going to test this soon.

furrtek, 2017-10-06:
This shouldn't make me smile...

ImDroided, 2017-11-03:
Anyone get to test this out yet? Did the brute force work? I still haven't made it out yet to try it.

ablanquart, 2017-11-03:
Yes, I've tested it out recently at a location with a TouchTunes "jukebox".  Lets just say, I should have read up on it a bit first.  I ended up powering it off and it took a while to boot back up.

notpike, 2017-11-07:
lol cool. I'll be testing it soon. Anyone else having the "Scan" crash at pin 205?

furrtek, 2017-11-08:
The scan crash at pin 205 or 206 is fixed in 196518457f61e820849ecd48a7b2c040fee3b463. Something was probably overflowing somewhere, stopping the baseband processor in between pins solved the problem.

chudgoo, 2017-12-23:
Just went through about an hour of scanning on a totally unpatched older unit and couldn't get a power off. Does the PIN on real units go beyond 255?

notpike, 2018-02-09:
Chudgoo,

Remember which juke box it was? some of the older models run on a different frequency.

chudgoo, 2018-02-09:
I’m fairly certain it was an “Allegro” unit. It looked older.