furrtek/portapack-havoc #99: POCSAG TX add sub-address field

stschuor, 2017-09-22:
Is it possible to separate the POCSAG sub-address from the message type?
Here in Switzerland, there's no correlation between them.
Fire fighting depts. often use them as severity levels (eg. A=severity 0.1, D=severity 2 however all of them are alphanumeric).
I managed to receive a test transmission from portapack after I reprogramed my pager to include subaddress D :-) (standard of my personal RIC is sub-address A alphanum)

furrtek, 2017-09-23:
Thanks for the suggestion :)
I added a "function" field in 9acfdcbd41a2d4c78a641bb584e286f5fa381391. The binary is up to date.
Choosing "Alphanumeric" as type will auto-set the function to D, but you can change it back to A manually afterwards.

stschuor, 2017-09-23:
Just tested the binary.
Works perfect on all four functions and also gets recognized as them on my pager :-)
Many thanks for the addition!

closed 2017-09-24 by furrtek