furrtek/portapack-havoc #106: ADSB Map is broken now

ImDroided, 2017-10-07:
When trying to load the map I get no world_map.bin file in /ADSB/ directory. I have the file on the SD card and it shows up in the file manager.

ImDroided, 2017-10-07:
same in transmit also

furrtek, 2017-10-14:
Fixed in 04c5b4d6070c72c693cb6f8d79e95c52c5faa7cd. Thanks :)

ImDroided, 2017-10-14:
Sweet. I will test later today and close this

closed 2017-10-14 by ImDroided

KR0SIV, 2019-02-07:
This is broken again it would appear, or at least the sdcard/adsb folder contains a .jpg file rather than a bin. At any rate it triggers the error, I renamed the ext but until more aircraft come into view i won't know for sure if it's loading it.

furrtek, 2019-02-07:
Did you follow https://github.com/furrtek/portapack-havoc/blob/master/sdcard/README.md ?

KR0SIV, 2019-02-07:
no, I'm sorry.. I feel like a damn fool now should've RTFM

furrtek, 2019-02-07:
No problem, information is spread across a lot of (too many) pages :)