furrtek/portapack-havoc #110: Transmitters > Soundboard > crash

Redouan29, 2017-10-15:
Build from source today. Cool new feature (last time I checked was a couple months ago).
But if you choose Transmitters > Soundboard , than the screen freezes and you forced to reset the device.

furrtek, 2017-10-28:
How many files do you have in the /WAV directory ?
Looks like scan_root_files isn't happy, might be a RAM issue like #109

chudgoo, 2017-10-30:
I too am seeing this. My SD card has about 10 files (all worked before upgrading to latest precompiled build)

Redouan29, 2017-11-06:
I cleaned out the whole SD card, but still no entrance to the menu/option "soundboard".

ImDroided, 2017-11-09:
Yep broke on my end also. I have to reset the unit upon clicking soundboard. I just compiled the latest

F0GPI, 2017-11-10:
Wave file viewer freeze portpack also after choosing the wave file in sdcard. the bug can possibly arise from the sdcard gestion eventually? (just an hypothesis).

furrtek, 2017-12-11:
RAM issue confirmed. Temporary fix for now is to get rid of the CTCSS option and limit files to 54 in 70c7646743ffc7af229bc8df8949b69feb23ecdc.

furrtek, 2018-05-15:
Fixed in 96828262d712b08abb675b7c4424fec47d8ffb93.