furrtek/portapack-havoc #111: A simple OOK remote calculator

Redouan29, 2017-10-15:
If will be nice to introduce a brute force method so you calculate the key of some OOK device at home. Very useful when you lost your remote of your on/off electricity switcher. Similar to im-me hack (Kamkar), if can be uses with De Bruin sequence.

furrtek, 2017-11-02:
The code's there, it still needs an UI.

Redouan29, 2017-11-06:
ok, clear thanks for the answer

fantuz, 2017-12-11:
Simple and *dual band* scan code based on know encoding-types is available here -> https://z4ziggy.wordpress.com/2014/06/27/rf-sniffer-open-gates-cars-and-rf-remote-controlled-devices-with-ease/ . RX only, but very easy to add a TX code with BF.

In the past in between different experiences, I gave a try to the mentioned "RCSwitch" library with immediate results, All the good catches are in there... at the moment (2017) is the most used and simplest OOK library for arduino. I just mention for the sake of parallel.

@furrtek apologize for the OT, I am just trying to stimulate collaboration and leave some written reference. I really hope other people (including me) will start pushing code so you will not be alone :+1: 

excellent work you doing

fantuz, 2017-12-24:
update on rc-switch lib to support HS2303 types -> https://github.com/sui77/rc-switch/pull/173#issuecomment-353771764

zwt2135, 2019-07-22:
Would love to see this. Currently having to listen and replay commands from a yard stick one onto my hackrf one. Would love to see this built in!

skfunky, 2019-11-17:
Its possible use It only with the hackrf?
How can i call this command?

thesle3p, 2020-01-28:
Is it known when we can expect this to be finished?

nickbouwhuis, 2020-01-29:
> Is it known when we can expect this to be finished?

never, unless you or someone else works on it and gives it back to the community