furrtek/portapack-havoc #112: Old M2K2 identifier

F4GEV, 2017-10-28:
Hello Mr,
i have try the radiosonde decoder with an old M2K2 (yes i have some M2K2 :-) )

His identifier is 0x648F23. your only identify M2K2 with 0x648F20 in your prog

@+
F4GEV

F4GEV, 2017-10-28:
Correction !!! I go my M2K2 outside !!
the M2K2 send when GPS position is determined
0x648F23 when not dermined 

@+

furrtek, 2017-10-30:
Added the identifier in 17b238f3a8a06ef10c7f9cebd435f7e82ec8464c, will try disabling the GPS module on a M10 to see if it also transmits ...23. Thanks :)

F4GEV, 2017-10-30:
Simple : plug the power supply to your M10, there is no backup battery for GPS.
the GPS take few minutes to lock with GPS (TTFF procedure)

Have fun
Laurent

closed 2017-11-08 by furrtek