furrtek/portapack-havoc #113: Radiosondes RX

F4GEV, 2017-10-29:
Hello,
the altitude decoded is wrong 
try this morning, displayed by portapack => -48m (yes minus)
decoded by software PC => 195m 

So the batterie voltage is wrong too displayed 578.000 mV  ????
My radiosond is powered wy variable power supply at 6,3 V

Bye

ImDroided, 2017-10-29:
I bet this is related to the issue with adsb decoding also.

furrtek, 2017-10-30:
I used my notes about the M10 only, apparently the M2K2 has diffent formulas for altitude and Vbatt. I'll add that later on today.

Giorgiofox, 2017-11-01:
I can't see data using a VAISALA RS41-SG
https://www.vaisala.com/en/products/instruments-sensors-and-other-measurement-devices/soundings-products/rs41

I've found one of this, and this is the model used here in Italy, can you please add support for this model?

Thanks
Giorgio

furrtek, 2017-11-02:
Will do :)

Redouan29, 2017-11-06:
I like to say that the radiosondes really works. Nice to see I have capture radiosonde :)

Maybe a noob question, can someone give a website or some converter so I can covert the latitude and Longitude to a Google maps? The map functionality is nice, but I can't zoom out.

ablanquart, 2017-11-07:
Redouan29,

You can put latitude and longitude directly into Google maps.

- ablanquart

Redouan29, 2017-11-07:
Ok thanks, it gives me a coordinate in China, can this be true? I mean I am almost 15.000 km away...

[image]

[image]

furrtek, 2017-11-08:
There are no checks of the frame's validity for now (except for the preamble trigger). You're getting an hex value for the signature which means the frame format isn't handled yet, so the GPS coordinates are garbage from incorrect decoding.
I'll make that clearer in the next update.

Redouan29, 2017-11-09:
@furrtek thanks dude! the software is great. I can't wait for the next update to track some real radio sondes :)

furrtek, 2017-11-10:
@Redouan29 Which type of sonde are you receiving ? I can't find anything about 0x2E3429.
@F4GEV Is your altitude really 195m ? Getting -48m means that the received altitude field is 0. Maybe you got a corrupt frame ?

jboone, 2017-11-10:
Wow, I need to get me some radiosondes and join in the fun. I have a box full of Viasala RS80-67s, but as far as I can tell, they're analog and have no GPS. Or maybe I can eBay some better ones...

Giorgiofox, 2017-12-07:
Sorry @furrtek: I will do recording of RS41-SG tomorrow, I've not read your message :-|
I can record IQ directly from HackRF or Spectravue with NetSDR.

Giorgiofox, 2017-12-07:
https://www.dropbox.com/s/03vdj2sf9m1p8cp/radiosonde%20RS41g%20recordings.zip?dl=0

fantuz, 2017-12-11:
Just a quick memorandum on some of the most used European kinds. Not inclusive but very detailed informations about signal deconding and corner-cases, along with images showing signal scope and frame infos.

--

J'ai retrouve ce site http://www.radiosonde.eu.bonplans.info/

Ici au chapitre 6, F5ZV detaille pas mal les RS80L, DFM-06, M2K2, M10, RS92KL, RS92SGP, SRS400-C34, RS92AGP et le logicielle de decodage SondeMonitor (trial 21 jours puis 25eur) cree par CT1EGC ex G3URZ.

D'ou "Le cas particulier du décodage des signaux d'une RS92AGP ou BGP". (c.f. http://www.radiosonde.eu.bonplans.info/RS06/RS06M.html et http://www.radiosonde.eu.bonplans.info/RS06/RS06H.html)

Je me demande si elles sont deja decodees par le HAVOC, apparement ces radiosondes survolent pas mal la Suisse et le centre Europe...

Giorgiofox, 2019-02-23:
Hello, 
any news about RS41-SG decoding?

Thanks in advance
Giorgio IZ2XBZ