furrtek/portapack-havoc #114: Bug on burger pager scan

F0GPI, 2017-10-30:
Hi Furrtek, 
i view one bug in the function burger pager.
If scan the library pager code and after few moment, the Portapack freeze. 
Pressed reset button and the Portapack reboot. 
it's normal or bug? 
Thanks.

fantuz, 2017-12-15:
Hello @F0GPI and @furrtek 
I would be interested in scanning burger pagers for prank !
But i never got to see these SCAN function working.: I infact thougt the button  was a placeholder for future dev.... yes i guess it was eventually causing freeze at times

As i forgot to mention it in another issue #134 I also report that SCAN is blocking with me too.
BTW having a SCAN function always sounds appealing, bravo!

jLynx, 2019-07-26:
Any update here?