furrtek/portapack-havoc #115: Capture - Replay not working - Very bad signal quality

chimera234, 2017-11-01:
Hello,
Actually don´t know if this is a bug or if I am doing something wrong here.

With "Receiver" --> "Audio" I am tuning into a strong Radio station, excellent signal quality
Afterwards I switch to "Capture" and record the signal for some time. From what I can tell file is properly written (BBDxxxx)
Once done I change to "Replay" and select on my radio the correct frequency for replay.
But I can hear the captured file only in a very bad quality. 
Actually don´t know if something went wrong during recording or during replay.

Many thanks in advance
Peter

furrtek, 2017-11-02:
Hello, it's indeed a known bug (there's a big list of bugs in firmware/main.cpp as comments which aren't posted as issues here yet), you're not doing anything wrong.
I still have to figure it out, it's either buffers getting mixed up or a sign problem somewhere. The recorded file should be good.

chimera234, 2017-11-02:
Thanks a lot for the clarification!
I will check to replay recorded file in hackrf mode this should work.

Pyr3x, 2017-12-04:
Any ETA on a fix? This is a feature I desperately need to be portable. Thank you! Also can I PayPal a donation? Where?

furrtek, 2017-12-04:
It's the next thing I need to get working, I've been putting it aside for too long. Will spend time on that tomorrow, hoping it'll be a matter of minutes and not hours.
If you really wish to donate, you can at furrtek@gmail.com.

Pyr3x, 2017-12-05:
Donation sent and thanks for the heads up on the ETA!

furrtek, 2017-12-05:
I spent most of the day trying to figure out what's going on.
There was a bug which caused the file buffers to be read backwards. That's fixed, but the main problem is still there: it seems that the buffers aren't filled completely or quickly enough. The replayed audio clearly shows buffer stalls.
I'm pretty sure it's not a performance issue since reading from SD cards should be faster than writing to them, and the baseband code has less to do than during recording... Took notes but nothing stands out for now.

furrtek, 2017-12-07:
Fixed in 1923c2bf1df8692cba0e8b670c2082c6a96fc797

chimera234, 2017-12-07:
Thank you so much for this!!!!!

fantuz, 2017-12-07:
donc pour te envoyer un don, on utilise ton addresse gmail en paypal ?
cimer pour les effort et la discussion toujours stimulante

furrtek, 2017-12-07:
Oui, si tu le souhaites :) Ravi que ça soit utile et d'enfin être arrivé à bout de ce bug.

closed 2018-04-18 by furrtek

x2plag, 2020-02-14:
Hello. This is still the issue for me.  I am using HAVOC portapack.
I am trying to record FM radio and replay it on the same freq.
When I replay recorded signal it has very very bad quality and I need to be in the very proximity to my FM reciver to actually improve signal quality. 
BUT when I do MIC transmition  on the same FM freq   it works flawlessly. When I press right button to activate MIC transmition I can hear only my voice from FM receiver. So signal strengh is very good. So I think there is no problem with hardware but something wrong with firmware.

I tried build firmware from the souce and use provided bin file - still have the same problem. 

Thank you! Need help very badly! :)