furrtek/portapack-havoc #116: Feature request  -  DJI drone intercept like mesmer

mwarchut, 2017-11-01:
Would it be possible to do something like this can.

https://department13.com/mesmer/

Essentially monitor the frequency range, identify the drone and then take control of it or interfere with it?  Or is this something way above the capability of the hackrf.

gh0stshell, 2017-11-01:
This would be awesome

furrtek, 2017-11-02:
DJI controls are done via standard WiFi, right ? I'm pretty sure that's a bit too much for the PortaPack to handle all alone :(

F4GEV, 2017-11-03:
I agree with furretek, so the mesmer solution use signals near 1 watt of power output. 
The card isn't able to deliver this output power to jam the signal, one external power amplifier by ISM band !!!!!

furrtek, 2017-11-03:
I believe they don't do jamming, but packet interception and modification.

ImDroided, 2017-11-03:
Is anyone aware of a amp for sale that would work for transmitting with the hackrf? And yes I believe they hijack the connection to the drone

mwarchut, 2017-11-07:
Bummer.There is a kid with a phantom  that keeps flying around my property.  I'd like to send it off on a flyaway.  heh

gh0stshell, 2017-11-07:
Don't know if this helps but my neighbor has a phantom while I own custom
drones that use a freq hopping protocol and was able to block his signal
and make it return to home on its own while nothing happened to my custom
ones, freq kept moving and could not use the scanner to get the signal as
the start and stop freq section is wider then what the scanner would allow
before the app crashed on me.

mwarchut, 2017-11-07:
So what you are saying is you were able to use the hackrf to jam the phantom?   Thats what I want to do. I have an amp and antenna I can set up  for the next time I see it out there.

ImDroided, 2017-11-07:
What are you guys using for a amp I need one to test with

ablanquart, 2017-11-08:
I believe what furrtek was getting at was that the processing power is too
much for the hackRF + portapack alone.  A hackRF + PC is a different
story..

mwarchut, 2017-11-08:
Agreed.   I may work on something with the hackrf connected to a pi or a laptop.

audioplatinum, 2017-11-30:
LNA4ALL ( http://lna4all.blogspot.hr ) can deliver over 100mW in TX mode (+21dBm) and at the same time use the HackRF Bias-T for DC power supplying the same amplifier.. I use this amp for testing, nice and small...