furrtek/portapack-havoc #117: MIC Transmit to SHURE Receivers?

Pyr3x, 2017-11-01:
This isn't exactly an issue per say but I was hoping to use the MIC Transmitter to send voice to my SHURE ULXP4 Receiver however, from my understanding it needs a "pilot tone" at 32 kHz...

Would it be possible to add this feature?

Reference on Pilot Tone: http://www.shure.com/americas/support/find-an-answer/disable-tone-key-on-ulx-receiver

furrtek, 2017-11-01:
No problem, I'll add a tone mixing feature in the following days.
I've heard that for Sennheiser receivers also.

Pyr3x, 2017-11-01:
Awesome! Thank You!

F4GEV, 2017-11-02:
I have sennheiser mic and receivers at my job, i can take one mic+rx for make tests if you want !

furrtek, 2017-11-09:
Added 19kHz, 28kHz, 32kHz, and 32.768kHz tone keys in 4465cfb9052d1dc84e1e0bca86436488bb571df2. "CTCSS" was renamed to "Tone key", they're in the end of the list after the CTCSS tones.
I'm not sure about the mixing ratio, I kept the CTCSS's 20% for now.

F4GEV, 2017-11-10:
I try to test this new version with EW 100 Rx and Mic this WE

Pyr3x, 2017-12-04:
I ran a test with my recivers at work, no luck. Any ideas?

furrtek, 2017-12-04:
Can you record and share a short IQ file of a transmission from one of the mics ?

Pyr3x, 2017-12-05:
I can! I will do that tomorrow!

Pyr3x, 2017-12-05:
I have the recordings here: https://1drv.ms/f/s!AhXrIQ5fw8Alnplynpkda4aTw3tmPA

furrtek, 2017-12-05:
Great, that was very helpful. I'm seeing the pilot tone at 32768Hz, have you tried one of the Sennheiser options ? Maybe I should remove the brand names.
It's a bit more quiet than expected but I guess the receiver doesn't care if the tone is too loud.
[image]

Pyr3x, 2017-12-05:
I will give it a try now! I had not thought of trying that...

Pyr3x, 2017-12-06:
The other options all failed, I get RF activity but it won’t let the audio through. Video links coming shortly.

Pyr3x, 2017-12-06:
https://youtu.be/96SBhmtJQDk

furrtek, 2017-12-06:
I'm seeing +/-38kHz modulation on the receiver's manual, maybe it doesn't like wrong bandwidth settings. I'll compare the spectrum sent by the PortaPack and your I/Q recordings, rather than just the audio.

Pyr3x, 2017-12-06:
Sorry for the video quality I was trying to record with one hand and change settings on the hackrf at the same time. The receiver model in that video is a ULXP4.

[image]

Pyr3x, 2017-12-12:
Do we still need testing here? If so what should be done for testing?