furrtek/portapack-havoc #118: issue compiling latest firmware

ImDroided, 2017-11-03:
Got this error after command make firmware

[ 18%] Building C object firmware/application/CMakeFiles/application.elf.dir/__/chibios-portapack/ext/fatfs/src/ff.c.obj
[ 18%] Building C object firmware/application/CMakeFiles/application.elf.dir/__/chibios-portapack/ext/fatfs/src/option/unicode.c.obj
[ 18%] Building CXX object firmware/application/CMakeFiles/application.elf.dir/main.cpp.obj
/home/twitch/SDR/portapack-havoc/firmware/application/main.cpp:62:1: error: version control conflict marker in file
 <<<<<<< ./firmware/application/main_LOCAL_5232.cpp
 ^~~~~~~
firmware/application/CMakeFiles/application.elf.dir/build.make:1712: recipe for target 'firmware/application/CMakeFiles/application.elf.dir/main.cpp.obj' failed
make[3]: *** [firmware/application/CMakeFiles/application.elf.dir/main.cpp.obj] Error 1
CMakeFiles/Makefile2:247: recipe for target 'firmware/application/CMakeFiles/application.elf.dir/all' failed
make[2]: *** [firmware/application/CMakeFiles/application.elf.dir/all] Error 2
CMakeFiles/Makefile2:103: recipe for target 'firmware/CMakeFiles/firmware.dir/rule' failed
make[1]: *** [firmware/CMakeFiles/firmware.dir/rule] Error 2
Makefile:118: recipe for target 'firmware' failed
make: *** [firmware] Error 2
twitch@strix:~/SDR/portapack-havoc/build$

F4GEV, 2017-11-03:
Hello, same for me

/portapack-havoc/firmware/application/main.cpp:62:1: error: version control conflict marker in file
 <<<<<<< ./firmware/application/main_LOCAL_5232.cpp
 ^~~~~~~
make[3]: *** [firmware/application/CMakeFiles/application.elf.dir/main.cpp.obj] Error 1
make[2]: *** [firmware/application/CMakeFiles/application.elf.dir/all] Error 2
make[1]: *** [firmware/CMakeFiles/firmware.dir/rule] Error 2
make: *** [firmware] Error 2

ImDroided, 2017-11-03:
Ok so until he fixes the file you can just delete the lines in firmware/application/main.cpp in the todo list section that start with 

<<<<<<<
>>>>>>>
|||||||
=======

Delete those lines and rebuild. 

The way it looks now....

Continuous (Fox-oring)
12s transmit, 48s space (Sprint 1/5th) 
60s transmit, 240s space (Classic 1/5 min) 
60s transmit, 360s space (Classic 1/7 min) 
*/
<<<<<<< ./firmware/application/main_LOCAL_5232.cpp
||||||| ./firmware/application/main_BASE_5232.cpp
//TODO: Use TransmitterView in TEDI/LCR, Numbers, ...
=======
//TODO: Use TransmitterView in TEDI/LCR, Numbers, ...
//TODO: Use TransmitterView in TEDI/LCR, Numbers, ...
>>>>>>> ./firmware/application/main_REMOTE_5232.cpp
//TODO: FreqMan: Remove and rename categories
<<<<<<< ./firmware/application/main_LOCAL_5232.cpp
||||||| ./firmware/application/main_BASE_5232.cpp
//TODO: Wav visualizer
//TODO: File browser view ?
=======
//TODO: Wav visualizer
//TODO: File browser view ?
//TODO: Wav visualizer
//TODO: File browser view ?
>>>>>>> ./firmware/application/main_REMOTE_5232.cpp
//TODO: Mousejack ?
//TODO: Move frequencykeypad from ui_receiver to ui_widget (used everywhere)
//TODO: ADS-B draw trajectory + GPS coordinates + scale, and playback
//TODO: RDS multiple groups (sequence)
//TODO: Use ModalMessageView confirmation for TX ?
//TODO: Use msgpack for settings, lists... on sd card

The way it should look to build without errors

Continuous (Fox-oring)
12s transmit, 48s space (Sprint 1/5th) 
60s transmit, 240s space (Classic 1/5 min) 
60s transmit, 360s space (Classic 1/7 min) 
*/
//TODO: Use TransmitterView in TEDI/LCR, Numbers, ...
//TODO: Use TransmitterView in TEDI/LCR, Numbers, ...
//TODO: Use TransmitterView in TEDI/LCR, Numbers, ...
//TODO: FreqMan: Remove and rename categories
//TODO: Wav visualizer
//TODO: File browser view ?
//TODO: Wav visualizer
//TODO: File browser view ?
//TODO: Wav visualizer
//TODO: File browser view ?
//TODO: Mousejack ?
//TODO: Move frequencykeypad from ui_receiver to ui_widget (used everywhere)
//TODO: ADS-B draw trajectory + GPS coordinates + scale, and playback
//TODO: RDS multiple groups (sequence)
//TODO: Use ModalMessageView confirmation for TX ?
//TODO: Use msgpack for settings, lists... on sd card

Redouan29, 2017-11-06:
indeed, if you delete or comment out those lines you can build from source again.

chudgoo, 2017-11-07:
I removed the offending lines to match the above, but am now getting:

portapack-havoc/firmware/application/ui_jammer.hpp:234:86: error: array must be initialized with a brace-enclosed initializer
  std::array<RangeView*, 3> range_views { &view_range_a, &view_range_b, &view_range_c };
                                                                                      ^
/Users/chud/portapack-havoc/firmware/application/ui_jammer.hpp:234:86: error: too many initializers for 'std::array<ui::RangeView*, 3u>'
make[3]: *** [firmware/application/CMakeFiles/application.elf.dir/ui_jammer.cpp.obj] Error 1
make[2]: *** [firmware/application/CMakeFiles/application.elf.dir/all] Error 2
make[1]: *** [firmware/CMakeFiles/firmware.dir/rule] Error 2
make: *** [firmware] Error 2

xmycroftx, 2017-11-08:
if you pull down the latest it should only have one merge conflict (in main.cpp)

furrtek, 2017-11-08:
Main.cpp cleaned up in 196518457f61e820849ecd48a7b2c040fee3b463. Sorry, I'm messy.

F4GEV, 2017-11-08:
Works with last release.

Thanks

ImDroided, 2017-11-09:
Ty

closed 2017-11-09 by ImDroided

chudgoo, 2017-11-09:
forgive the ignorance please... but i've tried compiling the latest on an ubuntu box and FOUR different macbooks (on 10.12, with homebrew) but on each I'm seeing error on 'make firmware'.

[ 35%] Building CXX object firmware/application/CMakeFiles/application.elf.dir/ui_jammer.cpp.obj
In file included from /Users/chud/temphavoc/portapack-havoc/firmware/application/ui_jammer.cpp:23:0:
/Users/chud/temphavoc/portapack-havoc/firmware/application/ui_jammer.hpp:234:86: error: array must be initialized with a brace-enclosed initializer
  std::array<RangeView*, 3> range_views { &view_range_a, &view_range_b, &view_range_c };
                                                                                      ^
/Users/chud/temphavoc/portapack-havoc/firmware/application/ui_jammer.hpp:234:86: error: too many initializers for 'std::array<ui::RangeView*, 3u>'
make[3]: *** [firmware/application/CMakeFiles/application.elf.dir/ui_jammer.cpp.obj] Error 1
make[2]: *** [firmware/application/CMakeFiles/application.elf.dir/all] Error 2
make[1]: *** [firmware/CMakeFiles/firmware.dir/rule] Error 2
make: *** [firmware] Error 2

furrtek, 2017-11-09:
You're probably using an outdated version of the toolchain, please use a recent one (2017) from https://developer.arm.com/open-source/gnu-toolchain/gnu-arm-embedded-toolchain/downloads
Let me know if you have troubles getting it to run.