furrtek/portapack-havoc #119: Alto Stealth Wireless

chudgoo, 2017-11-03:
Along the lines of #117 Do you think it would be possible to use the soundboard function with the Alto Stealth Wireless receivers?  This also appears to use a pilot tone, but I am unable to find the specific hz value...  If so, do you think dual channel (left/right) transmission would be possible so you could hit both speakers?

https://www.altoproaudio.com/products/stealth-wireless

https://www.altoproaudio.com/downloads/Stealth_Frequency_Chart_OUT.jpg

furrtek, 2017-11-09:
All pairs of channels seem to fit in a 2MHz span at most, I guess transmitting on both at once can be done if I figure out translating filters :)

chudgoo, 2017-11-10:
I spotted some of these attached to some huge PA speakers during a Halloween event and thought the 'Howie Scream' or christmas music wouldn't have been fun to interject.  ;)
https://www.youtube.com/watch?v=MqwuksvDFAQ